chipweinberger/flutter_blue_plus #666: Turn on and off Bluetooth search multiple times[Bug]:

anjiajie, 2023-11-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.27.5

### Flutter Version

3.13.7

### What OS?

Android

### OS Version

Android

### Bluetooth Module

unknown

### What is your problem?

On an android device, when I turn Bluetooth on and off multiple times, and then turn Bluetooth on again, FlutterBluePlus.scanResults.listen does not return any Result

### Logs

```shell
D/[FBP-Android](21737): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](21737): [FBP] onMethodCall: startScan
I/BluetoothAdapter(21737): isLeEnabled(): ON
D/BluetoothLeScanner(21737): onScannerRegistered() - status=6 scannerId=-1 mScannerId=0
```

chipweinberger, 2023-11-09:
try 1.27.6

anjiajie, 2023-11-09:
> try 1.27.6

That doesn't work, still the same problem

chipweinberger, 2023-11-09:
afaict, this is a bug in android.

try the example app

anjiajie, 2023-11-10:
> try the example app
> 
> maybe this is a bug in android.

Thanks you,I tried the example and the same problem occurred. It may really be an Android bug.

chipweinberger, 2023-11-10:
can you give more details about the phone & specific android version you are using

spehj, 2023-11-10:
Android has a limitation of 5 scans in 30 seconds. Maybe you broke this rule and this is not really an issue:

> In Android 7.0 and newer versions there is protection against Bluetooth LE scanning abuse. If your app starts and stops Bluetooth LE scans more than 5 times within 30 seconds, scan results will not be received temporarily
[source](https://www.silabs.com/documents/public/application-notes/an1200-1-bluetooth-mesh-2x-for-android-and-ios-adk.pdf)

chipweinberger, 2023-11-11:
@spehj , that a good point. @anjiajie are you scanning as soon as the adapter is enabled?

chipweinberger, 2023-11-11:
@spehj , that's a good point, but in the example app we do not break this rule. scanning is an explicit user action.

spehj, 2023-11-11:
@chipweinberger you are correct. As I understand @anjiajie, he just started scanning too many times in 30 seconds. This is a user action issue, not a flutter_blue_plus issue.

chipweinberger, 2023-11-27:
closing due to inactivity.

please reopen a new issue if there are new details.

closed 2023-11-27 by chipweinberger