chipweinberger/flutter_blue_plus #668: [Feature]: withServices support wildcard

keskink, 2023-11-09:
### FlutterBluePlus Version

1.27.6

### Flutter Version

3.13.9

### What OS?

All

### OS Version

all

### Bluetooth Module

ti 2460

### What is your feature request?

iOS getting servisUUID as "9635" and android getting as "00009635-0000-1000-8000-00805f9b34fb"
I should set two different uuid for one service uuid.

When scan method return Stream<ScanResult>, I was using as like 
.where((event) => event.advertisementData.serviceUuids.join(",").contains("9635")

Now scan method not supporting.

it would be nice, if we can add serviciUUID string as like "*9635*"

### Logs

```shell
n/a
```

chipweinberger, 2023-11-09:
thanks for reporting.

you can still manually filter just like before btw. You can use `oneByOne` mode.

chipweinberger, 2023-11-09:
btw, can you open a PR for this feature?

1. add 32-bit, 64-bit, support to Guid
2. remove mac support form Guid, it's never used
3. fix the issue of android returning long UUID. it should return the as-is UUID, in whatever length is advertised

chipweinberger, 2023-11-09:
fixed in 1.28.0

closed 2023-11-09 by chipweinberger