chipweinberger/flutter_blue_plus #670: [Bug]: Scan with services on Android throws "Invalid UUID string"

biklas7, 2023-11-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.28.7

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Asus Zenfone 9

### What is your problem?

Trying to scan devices with the following service:
`Guid('00001808-0000-1000-8000-00805f9b34fb')`

Throws `java.lang.IllegalArgumentException: Invalid UUID string: 1808`

Judging from the latest updates this is probably something that was worked on recently.
I'm pretty sure this was working before.

This is the debug log from start scan which shows it's using the wrong service id:
`I/flutter (21379): [FBP] <startScan> args: {with_services: [1808], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 4, android_scan_mode: 2, android_uses_fine_location: false}
`

### Logs

```shell
I/flutter (21379): [FBP] <startScan> args: {with_services: [1808], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 4, android_scan_mode: 2, android_uses_fine_location: false}

"java.lang.IllegalArgumentException: Invalid UUID string: 1808"
```

cBiscuitSurprise, 2023-11-10:
I'm seeing the same issue. I think it broken when `Guid.toString()` was updated to return `uuid` instead of `uuid128`. With the latest example app, if I add `withServices`:

```
[FBP] onMethodCall: startScan
D/AndroidRuntime(30034): Shutting down VM
E/AndroidRuntime(30034): FATAL EXCEPTION: main
E/AndroidRuntime(30034): Process: com.lib.flutter_blue_plus_example, PID: 30034
E/AndroidRuntime(30034): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1453, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity}: java.lang.IllegalArgumentException: Invalid UUID string: 1827
E/AndroidRuntime(30034):        at android.app.ActivityThread.deliverResults(ActivityThread.java:5527)
E/AndroidRuntime(30034):        at android.app.ActivityThread.handleSendResult(ActivityThread.java:5566)
E/AndroidRuntime(30034):        at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67)
E/AndroidRuntime(30034):        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
E/AndroidRuntime(30034):        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
E/AndroidRuntime(30034):        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
E/AndroidRuntime(30034):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2443)
E/AndroidRuntime(30034):        at android.os.Handler.dispatchMessage(Handler.java:106)
E/AndroidRuntime(30034):        at android.os.Looper.loopOnce(Looper.java:205)
E/AndroidRuntime(30034):        at android.os.Looper.loop(Looper.java:294)
E/AndroidRuntime(30034):        at android.app.ActivityThread.main(ActivityThread.java:8177)
E/AndroidRuntime(30034):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(30034):        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
E/AndroidRuntime(30034):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
E/AndroidRuntime(30034): Caused by: java.lang.IllegalArgumentException: Invalid UUID string: 1827
E/AndroidRuntime(30034):        at java.util.UUID.fromStringJava8(UUID.java:276)
E/AndroidRuntime(30034):        at java.util.UUID.fromString(UUID.java:223)
E/AndroidRuntime(30034):        at android.os.ParcelUuid.fromString(ParcelUuid.java:55)
E/AndroidRuntime(30034):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$3$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:520)
E/AndroidRuntime(30034):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda5.op(Unknown Source:22)
E/AndroidRuntime(30034):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.askPermission(FlutterBluePlusPlugin.java:1457)
E/AndroidRuntime(30034):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$askPermission$6$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:1470)
E/AndroidRuntime(30034):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda1.op(Unknown Source:6)
E/AndroidRuntime(30034):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onRequestPermissionsResult(FlutterBluePlusPlugin.java:1426)
E/AndroidRuntime(30034):        at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:785)
E/AndroidRuntime(30034):        at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:413)
E/AndroidRuntime(30034):        at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:816)
E/AndroidRuntime(30034):        at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:952)
E/AndroidRuntime(30034):        at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:9089)
E/AndroidRuntime(30034):        at android.app.Activity.dispatchActivityResult(Activity.java:8947)
E/AndroidRuntime(30034):        at android.app.ActivityThread.deliverResults(ActivityThread.java:5520)
E/AndroidRuntime(30034):        ... 13 more
```

```dart
// lib/screens/scan_screen.dart
...
Guid _guidMeshProvisioningService = Guid("00001827-0000-1000-8000-00805f9b34fb");
Guid _guidMeshProxyService = Guid("00001828-0000-1000-8000-00805f9b34fb");
...
Future onScanPressed() async {
    try {
      // android is slow when asking for all advertisments,
      // so instead we only ask for 1/8 of them
      int divisor = Platform.isAndroid ? 8 : 1;
      await FlutterBluePlus.startScan(
        timeout: const Duration(seconds: 15),
        continuousUpdates: true,
        continuousDivisor: divisor,
        withServices: [
          _guidMeshProvisioningService,
          _guidMeshProvisioningService,
        ],
      );
...
```

cBiscuitSurprise, 2023-11-10:
I created #671 to fix this specific case, but I would guess there are other areas that this particular issue impacts (I just don't have a mature enough setup to test all the other aspects of BLE yet).

chipweinberger, 2023-11-10:
try master branch

chipweinberger, 2023-11-10:
fixed in 1.28.8

closed 2023-11-10 by chipweinberger