chipweinberger/flutter_blue_plus #674: [Help]: is connect without scan working android 10 and below.

keskink, 2023-11-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.28.0

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

android 10

### Bluetooth Module

Huawei Mate 20 Lite

### What is your problem?

   as far as I understand, we can not connect w/o scan before android 11.
   I have android 10 (Huawei mate 20 lite) phone. 
   I can not able to connect w/o scan and autoconnect not working as well.

Does anyone have experience or suggestions for a solution to this issue?

### Logs

```shell
n/a
```

chipweinberger, 2023-11-11:
In general, I don't recommend trying to connect without scanning.

In BLE, for the first time, scanning is *required* before you can connect. Your phone must get the connection parameters, which can only happen during scan.

After the first scan, you can connect without scanning again. 

As an exception to this rule, if the device is saved in the system setting (bonded), you may be able to connect without scanning again.

But you will always need to scan at least once. 

--- 
Also, I don't recommend using autoconnect. 1. is does not work on iOS. 2. it's pretty buggy and there are a lot of corner cases that cause it to not work. 

It's easier to just listen for disconnection & then call connect again.

keskink, 2023-11-12:
Sorry for this. I want to make a correction. After some attempts, I was able to connect without scan.
Dear [chipweinberger], Many thanks for your answer.  I agree with everything you wrote. 
in my application, I am adding ble device to my "known devices" list with scan. After then, I am expecting to connect w/o scan.
Anyway, I can able to connect w/o scan. (there is no any issue all ios version, android version 11 and after.) 
But autoconnect function is very important for my project.
You wrote that autoconnect does not work on iOS. I don't agree this. iOS always supports autoconnect. There is no any parameters autoconnect on/off. But it is working.
in my application:
I am calling connect method for all devices in known devices list when app starts.
When devices power on or became to ble coverage area, I am expecting to connect.
it is working in iOS and android v11 and after versions.
I just need to fix for android 10 and before.
Thanks again

chipweinberger, 2023-11-12:
are you using ios 17?

I added support for autoconnect for ios 17 but haven't tested it yet.

maybe it works!

keskink, 2023-11-12:
No, I am not using 17 yet. But I don't have any issue with iOS.
Autoconnect working in iOS for me. I meant;
I am scanning my device first. Then I am adding to my known devices list. Then I am calling .connect() method.
not important device powered on or off. if my device powered on, it is connecting automatically. Or when my device enter to coverage area it is connecting automatically. This is working in iOS w/o any problem. 
in android (>= v11), the connect method waiting connection some period, then going to timeout exception. I am calling connect method again. it is working for android v11 and after.
But in android v10 and before, connect method do not connect, if device powered on after call connect method. 
I hope I can explain my issue.

chipweinberger, 2023-11-13:
yes it makes sense what you are saying but find it surprising

- why does ios not timeout?
- why does ios do "autoconnect" even without the new ios 17 autoconnect flag

keskink, 2023-11-13:
As far as I read some documents. iOS has connectionTimeout but I am using default value. Not sure but, I think it is 30 sec. (30 sec is enough for me)
I am calling connect method when device disconnected or ble has any connection error.
and yes autoconnect working in ios for long time. may be they add autoconnect parameter for cancel autoconnect after iOS 17.+
in androd v11.+ is working as like ios.
but not working before  v10. I can give timeout. But it does not make sense due to, autoconnect not working during timeout period.

chipweinberger, 2023-11-13:
"simply call the connect-method again in func centralManager(_ central: CBCentralManager, didDisconnectPeripheral peripheral: CBPeripheral, error: Error?) because connection requests do not time out, the iOS device will reconnect when the user returns home."

wow you are right. iOS will never time out. interesting!

https://medium.com/arkulpa/ios-stay-connected-to-an-external-ble-device-as-much-as-possible-699d434846d2

chipweinberger, 2023-11-13:
this is the new ios 17 option

https://developer.apple.com/documentation/corebluetooth/cbconnectperipheraloptionenableautoreconnect

chipweinberger, 2023-11-27:
closing this due to inactivity