chipweinberger/flutter_blue_plus #676: [Bug]: Write only works once or twice then it gets stuck

iKK001, 2023-11-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.28.7

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

iOS 17.1.1

### Bluetooth Module

ESP32-S3

### What is your problem?

I have problems with the "write" function. (i.e. sending bytes from Flutter-App to ESP32-S3 Arduino)

I do byte-"writes" from the iOS Flutter App (using flutter_blue_plus) to the ESP32-S3 Arduino.

After connecting, discoverServices, get characteristic, everything seems to work: i.e. I can always write one Byte or two.

But as soon as I write more, then the following error crops up:

```
flutter: Write Error: FlutterBluePlusException | writeCharacteristic | fbp-code: 1 | Timed out after 15s
```

It seems that there is no "acknowledge" somehow from the ESP-S3.

I get this error also with the Example-App !!!

My Arduino code looks like this:
(only most important parts..)

```
#define BLE_POLL_INTERVALL  50
uint64_t previousMillis = 0;
uint64_t currentMillis = 0;

BLEService teddyDataService(BLE_UUID_TEDDY_SERVICE);
BLECharacteristic teddyCharacteristic(BLE_UUID_TEDDY_CHARACTERISTIC, BLERead | BLEWrite | BLENotify, sizeof teddyData.bytes);

union teddy_data_struct {
    struct __attribute__((packed)) {
        byte values[NUMBER_OF_SENSORS];
    };
    uint8_t bytes[NUMBER_OF_SENSORS * sizeof(byte)];
};

union teddy_data_struct teddyData;

void loop() {

    // listen for BLE central to connect:
    BLEDevice central = BLE.central();
[... 245 lines cut ...]
      print("Start Scan Error: $e");
    }
  }

  void connectDevice(BluetoothDevice device) {
    device.connectAndUpdateStream().catchError((e) {
      print("Error connecting: $e");
    });
  }
}
```

### Logs

```shell
flutter: Teddy Device Found
flutter: Discover Services: Success
flutter: Teddy Service Found
flutter: Teddy Characteristic Found
flutter: Write Success
flutter: [127]
flutter: Write Success
flutter: [104]
flutter: Write Error: FlutterBluePlusException | writeCharacteristic | fbp-code: 1 | Timed out after 15s
```

iKK001, 2023-11-13:
I figured it out:  it was the `if (central.rssi() != 0) {...}` check that I had in my Arduino code that caused the "write" issue !

Simply taking the if/then out of my code made everything work smoothly.

chipweinberger, 2023-11-13:
i'm not sure. it could be a bug in the arruino code.

closed 2023-11-13 by chipweinberger