chipweinberger/flutter_blue_plus #678: [Bug]: Unable to disconnect cached connection

mengzhidaren, 2023-11-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.28.10

### Flutter Version

Flutter 3.13.9

### What OS?

Android

### OS Version

android9

### Bluetooth Module

none

### What is your problem?

Question 1: 
 final cache = await FlutterBluePlus.systemDevices
 for (var element in cache) {
 //await element.disconnect(queue: true);
 await element.disconnect(queue: false);
}
Unable to disconnect cached connection

Question 2:
If connecting a cache device:
 final cache = await FlutterBluePlus.systemDevices
 for (var element in cache) {
element .connect()......
//....
await characteristic.setNotifyValue(true)
characteristic.onValueReceived.listen(e){
//There will be duplicate callback data here
//The device's callback will send the same data twice each time
}
}

### Logs

```shell
no
```

chipweinberger, 2023-11-13:
for question 1. you are correct. Those devices are not connected to your app. you cannot disconnect them. You can try connecting then disconnecting. 

for question 2, see README. duplicate data is common problem.

closed 2023-11-13 by chipweinberger

mengzhidaren, 2023-11-14:
Thank you so much