chipweinberger/flutter_blue_plus #680: [Bug]: Error when scanning on Android using withServices

etx, 2023-11-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.28.10

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

MacOS 14.1.1

### Bluetooth Module

Blackmagic Pocket Camera 

### What is your problem?

Only on Android devices, iOS and MacOS is working as expected.

When I start a scan with a list of services to scan for like this:
`FlutterBluePlus.startScan(timeout: const Duration(seconds: 15), withServices: _blackmagicServiceList)`

I see this error in the console:
`E/flutter (18870): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: type 'Null' is not a subtype of type 'int'
E/flutter (18870): #0      new BmScanAdvertisement.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:156:17)
E/flutter (18870): #1      new BmScanResponse.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:177:46)
E/flutter (18870): #2      FlutterBluePlus.startScan.<anonymous closure> (package:flutter_blue_plus/src/flutter_blue_plus.dart:217:39)
E/flutter (18870): #3      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
E/flutter (18870): #4      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (18870): #5      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)`

It looks like the rssi isn't populated when creating a new `BmScanAdvertisement` instance?

If I remove the withServices property from the startScan call everything works as I would expect.

### Logs

```shell
I/flutter (18870): Bluetooth state change: BluetoothAdapterState.on
V/DartMessenger(18870): Received message from Dart over channel 'flutter/platform'
V/DartMessenger(18870): Deferring to registered handler to process message.
V/PlatformChannel(18870): Received 'SystemSound.play' message.
I/flutter (18870): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (18870): │ #0   Scan.start (package:pocket_app/core/providers/devicesProvider.dart:43:9)
I/flutter (18870): │ #1   ConnectView.build.<anonymous closure> (package:pocket_app/core/views/connect_view.dart:117:63)
I/flutter (18870): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (18870): │ 💡 DevicesProvider Scan
I/flutter (18870): └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
V/DartMessenger(18870): Received message from Dart over channel 'flutter_blue_plus/methods'
V/DartMessenger(18870): Deferring to registered handler to process message.
D/[FBP-Android](18870): [FBP] onMethodCall: startScan
D/BluetoothAdapter(18870): isLeEnabled(): ON
D/BluetoothLeScanner(18870): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
I/flutter (18870): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (18870): │ #0   Scan.start.<anonymous closure> (package:pocket_app/core/providers/devicesProvider.dart:50:15)
I/flutter (18870): │ #1   <asynchronous suspension>
I/flutter (18870): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (18870): │ 💡 Scan complete
I/flutter (18870): └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
V/DartMessenger(18870): Sending message with callback over channel 'flutter_blue_plus/methods'
V/DartMessenger(18870): Received message reply from Dart.
E/flutter (18870): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: type 'Null' is not a subtype of type 'int'
E/flutter (18870): #0      new BmScanAdvertisement.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:156:17)
E/flutter (18870): #1      new BmScanResponse.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:177:46)
E/flutter (18870): #2      FlutterBluePlus.startScan.<anonymous closure> (package:flutter_blue_plus/src/flutter_blue_plus.dart:217:39)
E/flutter (18870): #3      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
E/flutter (18870): #4      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (18870): #5      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (18870): #6      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (18870): #7      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (18870): #8      _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (18870): #9      _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
E/flutter (18870): #10     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (18870): #11     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (18870): #12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (18870): #13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (18870): #14     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (18870): #15     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
E/flutter (18870): #16     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (18870): #17     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (18870): #18     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (18870): #19     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
E/flutter (18870): #20     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
E/flutter (18870): #21     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
E/flutter (18870): #22     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter (18870): #23     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
E/flutter (18870): 
V/DartMessenger(18870): Received message from Dart over channel 'flutter_blue_plus/methods'
V/DartMessenger(18870): Deferring to registered handler to process message.
D/[FBP-Android](18870): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(18870): isLeEnabled(): ON
Application finished.
```

chipweinberger, 2023-11-13:
fixed 1.28.12

closed 2023-11-13 by chipweinberger

etx, 2023-11-13:
Wow, thanks! Moved from flutter_blue to flutter_blue_plus today in my app. I must say, Nice work!

chipweinberger, 2023-11-13:
love to hear it! how was the migration?

etx, 2023-11-13:
> love to hear it! how was the migration?

Easy peasy thanks to the docs and deprecation annotations.