chipweinberger/flutter_blue_plus #681: [Bug]: When calling startScan during the cold start of the application, it results in an error, even though Bluetooth is already enabled.

SmallBlueWhale, 2023-11-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.28.10

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

ios 17.0.1

### Bluetooth Module

unknown

### What is your problem?

When calling startScan during the cold start of the application, it results in an error, even though Bluetooth is already enabled. This issue does not occur during a hot start. As logs shown:

### Logs

```shell
Unhandled Exception: PlatformException(startScan, bluetooth must be turned on, null, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
<asynchronous suspension>
#2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:464:13)
<asynchronous suspension>
#3      FlutterBluePlus.startScan (package:flutter_blue_plus/src/flutter_blue_plus.dart:223:5)
<asynchronous suspension>
```

SmallBlueWhale, 2023-11-14:
Adding a note: This issue only occurs during the cold start of the application. If the method is called after the application has been running for some time, no problem is observed. Even the FlutterBluePlus.isOn method has issues during cold start.

TinyGiantDigital, 2023-11-14:
+1 for this issue.
It seems to be iOS only. If I re-run startScan 1 second later it works as intended.

chipweinberger, 2023-11-14:
iOS takes awhile to initialize bluetooth

you neeed to wait for adapterState to turn on.

closed 2023-11-14 by chipweinberger

samir4055, 2024-11-15:
> iOS takes awhile to initialize bluetooth
> 
> you need to wait for `adapterState` to turn on.
> 
> **Simplest fix:**
> 
> ```
> // wait for bluetooth to turn on & permission granted
> await FlutterBluePlus.adapterState.where((val) => val == BluetoothAdapterState.on).first;
> 
> // start scanning
> // ....
> ```

I use this, but it does not work for me

mgeilich, 2025-06-21:
worked for me, thanks.