chipweinberger/flutter_blue_plus #684: [Bug]: App crashes when starting device scan on Android 10

sdgroot, 2023-11-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.28.13

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

Android 10

### Bluetooth Module

Not sure

### What is your problem?

On a Motorola g8 plus with Android 10, when I start a device discovery scan with the keywords param, the app crashes. Without this parameter it works fine. Also, on a Pixel 7a with Android 13 works fine.

```dart
try {
  await FlutterBluePlus.startScan(withKeywords: ['_HLM']);
} catch (error, trace) {
  Fimber.e('Error starting device discovery', ex: error, stacktrace: trace);
}
```

Thanks for this excellent library btw! Keep up the good work, it's much appreciated!

### Logs

```shell
D/[FBP-Android](23093): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](23093): [FBP] initializing BluetoothAdapter
D/[FBP-Android](23093): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](23093): [FBP] connectedPeripherals: 0
D/[FBP-Android](23093): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](23093): [FBP] onMethodCall: startScan
E/AndroidRuntime(23093): FATAL EXCEPTION: main
E/AndroidRuntime(23093): Process: [redacted], PID: 23093
E/AndroidRuntime(23093): java.lang.NoSuchMethodError: No virtual method setAdvertisingDataType(I)Landroid/bluetooth/le/ScanFilter$Builder; in class Landroid/bluetooth/le/ScanFilter$Builder; or its super classes (declaration of 'android.bluetooth.le.ScanFilter$Builder' appears in /system/framework/framework.jar)
E/AndroidRuntime(23093): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$3$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:548)
E/AndroidRuntime(23093): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda5.op(Unknown Source:24)
E/AndroidRuntime(23093): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissions(FlutterBluePlusPlugin.java:1462)
E/AndroidRuntime(23093): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:479)
E/AndroidRuntime(23093): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/AndroidRuntime(23093): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/AndroidRuntime(23093): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/AndroidRuntime(23093): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/AndroidRuntime(23093): 	at android.os.Handler.handleCallback(Handler.java:883)
E/AndroidRuntime(23093): 	at android.os.Handler.dispatchMessage(Handler.java:100)
E/AndroidRuntime(23093): 	at android.os.Looper.loop(Looper.java:241)
E/AndroidRuntime(23093): 	at android.app.ActivityThread.main(ActivityThread.java:7582)
E/AndroidRuntime(23093): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(23093): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/AndroidRuntime(23093): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:941)
```

chipweinberger, 2023-11-15:
ah, you are right. This `setAdvertisingDataType` requires minSdk33

I will add a check for it. Feel free to open a PR. It should be pretty obvious by looking at the other code.

chipweinberger, 2023-11-15:
fixed 1.28.14

closed 2023-11-15 by chipweinberger