chipweinberger/flutter_blue_plus #685: Fast Data transfer from BLE device.

karthikmohan4, 2023-11-15:
As read & write method is async. It would ideally take 54sec to transfer 9654 Bytes ,  packet by packet from the BLE device.   If  MTU is set more than 23, there is data loss. 

How to basically achieve the fast data transfer in less time. 

BLE device version : 4.2

chipweinberger, 2023-11-15:
increase mtu.

if there is data loss it is a bug in your code or device.

joyhope, 2023-11-16:
The library is no problem for fast BLE data transfer.  Thanks to the author's hard working. 

If problems happen, it is the device and your code.

karthikmohan4, 2023-11-16:
So what should be the ideal time to receive the data from BLE device for ex: 9654 bytes.?

Apart from read/write does any method helps us to retrieve data fast like sstream. 

My code follows (req-res) method. 
1. write some data 
2. read the data (takes 2 sec to read 517 Bytes).

Also the ble device version is 4.2.   Would that be  a reason for slow transfer. ?

Can you provide any resource related to this. 

BTW thanks for the package.

joyhope, 2023-11-16:
In fact, it is a little complicated.  There is a negotiation between PHONE (BLE SERVER) and the device for the interval, MTU, and some other communication parameters.  If you could control your device's BLE code, and then you could adjust BLE data performance, otherwise nothing could be improved.  These parameters are much more important than the BLE library.

chipweinberger, 2023-11-16:
it will be faster if your ble device supports notifications & you use `setNotifyValue` and `onValueReceived`.

chipweinberger, 2023-11-17:
closing because there is nothing more for FBP to do

closed 2023-11-17 by chipweinberger