chipweinberger/flutter_blue_plus #686: [Help]: advName and platformName confusion

iKK001, 2023-11-15:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.28.7

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

17.1.1

### Bluetooth Module

ESP32-S3 Wroom 1U

### What is your problem?

I have 10 ESP32-S3 PCB boards available.

Two of them work as expected (i.e. Bluetooth wonderfully works and tested with Flutter Example App).

I have all 10 modules programmed the very same way (i.e. identical Arduino code, identical Flutter code).

However 8 of them do not work yet and I made the following observation :

The 8 modules show `platformName` equal to `advName` for some odd reason - even tough, I expect them to have different names (since the two working modules do show these names distinct as expeected).

Here are code examples :

The Arduino code says

```
#define MYAPP_ID                                      "Sr2dyz6ubvogXDgM1alU"
#define PERIPHERAL_LOCAL_NAME       "MyApp 001"

BLE.setLocalName(MYAPP_ID);    // expected to be advName
BLE.setDeviceName(PERIPHERAL_LOCAL_NAME);    // exptected to be platformName
```

The Flutter code says :

```
_scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      List<String> advNameArr = results.map((d) => d.device.advName).toList();
      List<String> platformNameArr = results.map((d) => d.device.platformName).toList();
      if (advNameArr.contains("Sr2dyz6ubvogXDgM1alU")) {
        print("advName found !");
      }
      if (platformNameArr.contains("MyApp 001")) {
        print("platformName found !");
      }
      if (platformNameArr.contains("Sr2dyz6ubvogXDgM1alU")) {
        print("hmmmm why are 8 modules showing this print message ????????");
      }
      // ...
}
```

At this point, I am not sure whether it is an issue of the Arduino-code or if it is a flutter_ble_plus problem ??

Any idea what could be wrong here ??

Why is advName and platformName the same for 8 modules but not for 2 modules ??? (having identical code throughout !)

### Logs

```shell
flutter: advName found !
flutter: hmmmm why are 8 modules showing this print message ????????
flutter: advName found !
flutter: hmmmm why are 8 modules showing this print message ????????
flutter: advName found !
flutter: hmmmm why are 8 modules showing this print message ????????
flutter: advName found !
flutter: hmmmm why are 8 modules showing this print message ????????
flutter: advName found !
flutter: hmmmm why are 8 modules showing this print message ????????
flutter: advName found !
flutter: hmmmm why are 8 modules showing this print message ????????
```

iKK001, 2023-11-15:
I made the problem-observation #686 also with different BLE scanning software and I had to observe that the deviceName was definitively wrong for all 8 false modules. So I think we can assume that it is rather an ArduinoBLE problem than a Flutter problem.

I therefore posted the issue also on the [ArduinoBLE forum here....](https://forum.arduino.cc/t/ble-setlocalname-and-ble-setdevicename-are-messed-up/1189526)

chipweinberger, 2023-11-15:
you need to connect first.

the gap name is a ble characteristic. it is not advertised. only the advertised name is advertised.

closed 2023-11-15 by chipweinberger

iKK001, 2023-11-15:
do I understand you correctly:

`BLE.setLocalName(...);` : This creates the advertised name and can be called in Arduiono-setup.

`BLE.setDeviceName(...);` : This can only be called after you have connected - it eventually will lead to the platformName.