chipweinberger/flutter_blue_plus #688: [Bug]:

EdwardRodr1guez, 2023-11-16:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.28.14

### Flutter Version

3.10.1

### What OS?

iOS

### OS Version

IOS 16

### Bluetooth Module

esp32

### What is your problem?

Hello, my ios device is getting closed its connection after a 30 seconds aprox being connected. Its a little curious because if I set a higher connection timeout when I use .connect() function, this takes also a higher time to occur the disconnection. When I dont specify a timeout it takes a longer time to disconnect. The point is that, the device is lossing its bluetooth connection. 

Additionally, this didnt happend in version 1.17.5 exactly which was the previous version I was using 🤔  

### Logs

```shell
[FBP-iOS] didDisconnectPeripheral: [Error] The connection has timed out unexpectedly.
flutter: [FBP] \^[[1;30m[[ OnConnectionStateChanged ]]\^[[0m result: \^[[1;33m{remote_id: CD955241-1F02-F1A9-B376-8A6757330184, connection_state: 0, disconnect_reason_code: 6, disconnect_reason_string: The connection has timed out unexpectedly<…>
```

chipweinberger, 2023-11-16:
you can google this error. its from iOS.

it means your device stopped working.

closed 2023-11-16 by chipweinberger

TanujW, 2023-12-29:
@chipweinberger What's the best way to handle this? The example app seems to be crashing when this happens/

chipweinberger, 2023-12-29:
catch the error, and reconnect

TanujW, 2023-12-29:
@chipweinberger Thank you!