chipweinberger/flutter_blue_plus #690: [Help]: No implementation found for method startScan

max1lock, 2023-11-17:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.28.14

### Flutter Version

3.13.6

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Samsung S22+

### What is your problem?

Hello,
I'm trying to add flutter blue plus to my project in order to retrieve the list of bluetooth devices from a phone.

I tried the example provided and as is it works fine.
When trying to integrate blue plus into my project I am faced with the error present in logs.
I've tried several code implementations (including the one given in the ReadMe) and I'm still facing the same type of error.

You will find the code here : [](https://github.com/PTT-Alcatel/frontend/blob/feature/24-flutter-blue-plus/lib/pages/bubbles_page.dart)

I would be very grateful for your help. 

Maxlock

### Logs

```shell
I/flutter (28940): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 4, android_scan_mode: 2, android_uses_fine_location: false}
E/flutter (28940): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method startScan on channel flutter_blue_plus/methods)
E/flutter (28940): #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:308:7)
E/flutter (28940): <asynchronous suspension>
E/flutter (28940): #1      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:469:13)
E/flutter (28940): <asynchronous suspension>
E/flutter (28940): #2      FlutterBluePlus.startScan (package:flutter_blue_plus/src/flutter_blue_plus.dart:223:5)
E/flutter (28940): <asynchronous suspension>
E/flutter (28940):
```

chipweinberger, 2023-11-17:
never seen this error before.

chipweinberger, 2023-11-19:
> If you are using hot restart or hot reload, it won't do the trick. Since Flutter has to inject plugin dependencies into the platform-specific parts of your app, hot restart/hot reload is not enough to trigger the injection.

You need to fully stop the app, then run, so that it recompiles and loads the native code.

closed 2023-11-19 by chipweinberger