chipweinberger/flutter_blue_plus #692: [Feature]:

ToxicGamer-exe, 2023-11-19:
### FlutterBluePlus Version

1.28.13

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

Not needed

### What is your feature request?

I'd like to get bool, that would indicate whether the user accepted or rejected the turnOn() dialog. Not only the func doesn't return anything, but if the user rejects, the awaited request is stuck unless the timeout goes out and then it throws exception (basically the only indication user didn't accept). Ending the timeout after the rejection would be appreciated ofc :D 

### Logs

```shell
No logs needed
```

chipweinberger, 2023-11-19:
if you want this, feel free to open a PR.

chipweinberger, 2023-11-20:
I've pushed a fix to master branch. try it out! untested.

ToxicGamer-exe, 2023-11-20:
I'm gonna test it right out, thanks in advance! :)

chipweinberger, 2023-11-20:
did it work?

chipweinberger, 2023-11-20:
I tested it, it works.

closed 2023-11-20 by chipweinberger

chipweinberger, 2023-11-20:
fixed in 1.29.5