chipweinberger/flutter_blue_plus #693: [Bug]: iOS BLE communication error

JLdevFr, 2023-11-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.29.4

### Flutter Version

3.13.9

### What OS?

iOS

### OS Version

17.1.1

### Bluetooth Module

Espressif esp32-c3 mini

### What is your problem?

Communication with iOS with certain BLE modules has stopped for some time. The UUIDs have been shortened, which does not pose any issues. However, with certain BLE modules, we are able to connect to the BLE module, read its services and characteristics, but when we attempt to communicate, we encounter a 'service not found' error. I am using ESPRESSIF BLE modules. It's worth noting that this issue does not arise with Microchip or Minew modules. Additionally, I have also tried with the example application with DART 3.1.5.

### Logs

```shell
[FBP-iOS] handleMethodCall: connect
flutter: [FBP] \^[[1;30m<connect>\^[[0m result: \^[[1;33mtr
[FBP-iOS] didConnectPeripheral
flutter: [FBP] \^[[1;30m[[ OnConnectionStateChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, connection_state: 1, disconnect_reason_code: null, disconnect_reason_string: nul
flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, success: 1, error_string: success, error_code: 0, mtu: 2
flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, success: 1, error_string: success, error_code: 0, mtu: 50
flutter: [FBP] \^[[1;30m<discoverServices>\^[[0m args: \^[[1;35mDD017D95-5AD9-E191-11A4-6A0358E5F0
[FBP-iOS] handleMethodCall: discoverServices
flutter: [FBP] \^[[1;30m<discoverServices>\^[[0m result: \^[[1;33mtr
[FBP-iOS] didDiscoverServices
[FBP-iOS] Found service: 0000FFF0-0000-1000-8000-00805F9B34FB
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
flutter: [FBP] \^[[1;30m[[ OnDiscoveredServices ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, success: 1, services: [{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, characteristics: [{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, descriptors: [{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, characteristic_uuid: 0000fff2-0000-1000-8000-00805f9b34fb, descriptor_uuid: 2902, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null}], characteristic_uuid: 0000fff2-0000-1000-8000-00805f9b34fb, properties: {notify: 1, indicate_encryption_required: 0, write: 1, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null}, {remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, descriptors: [], characteristic_uuid: 0000fff1-0000-1000-8000-00805f9b34fb, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 1, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null}], is_primary: true, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, included_services: []}], error_string: success, error_code:
flutter: [FBP] \^[[1;30m<writeCharacteristic>\^[[0m args: \^[[1;35m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, service_uuid: fff0, secondary_service_uuid: null, characteristic_uuid: fff2, write_type: 0, allow_long_write: 0, value: 6f80ea7
[FBP-iOS] handleMethodCall: writeCharacteristic
flutter: [FBP] \^[[1;30m<readCharacteristic>\^[[0m args: \^[[1;35m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, service_uuid: fff0, secondary_service_uuid: null, characteristic_uuid: fff
[FBP-iOS] handleMethodCall: readCharacteristic
flutter: [FBP] \^[[1;30m<disconnect>\^[[0m args: \^[[1;35mDD017D95-5AD9-E191-11A4-6A0358E5F0
[FBP-iOS] handleMethodCall: disconnect
[FBP-iOS] disconnect: canceling connection in progress
flutter: [FBP] \^[[1;30m<disconnect>\^[[0m result: \^[[1;33mtr
[FBP-iOS] didDisconnectPeripheral
flutter: [FBP] \^[[1;30m[[ OnConnectionStateChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, connection_state: 0, disconnect_reason_code: 23789258, disconnect_reason_string: connection cancele
```

chipweinberger, 2023-11-20:
I dont see any errors in your logs?

JLdevFr, 2023-11-20:
with the example code, the error appear on the screen with a snackbar.
On my app, the try cath return error since the last update of FBP : 
[FBP-iOS] handleMethodCall: writeCharacteristic
flutter: communication error: PlatformException(writeCharacteristic, service not found 'fff0', null, null)

chipweinberger, 2023-11-20:
try adding logs to this. 

```
- (CBService *)getServiceFromArray:(NSString *)uuid array:(NSArray<CBService *> *)array
{
    for (CBService *s in array)
    {
       NSLog(@"%@ %@", [s.UUID uuidStr], uuid); // add this log
        if ([[s.UUID uuidStr] isEqualToString:uuid])
        {
            return s;
        }
    }
    return nil;
}
```

chipweinberger, 2023-11-20:
i think these three functions need to be this.

you can try it and let me know.

```
- (CBService *)getServiceFromArray:(NSString *)uuid array:(NSArray<CBService *> *)array
{
    for (CBService *s in array)
    {
        if ([s.UUID isEqual:[CBUUID UUIDWithString:uuid]])
        {
            return s;
        }
    }
    return nil;
}

- (CBCharacteristic *)getCharacteristicFromArray:(NSString *)uuid array:(NSArray<CBCharacteristic *> *)array
{
    for (CBCharacteristic *c in array)
    {
        if ([c.UUID isEqual:[CBUUID UUIDWithString:uuid]])
        {
            return c;
        }
    }
    return nil;
}

- (CBDescriptor *)getDescriptorFromArray:(NSString *)uuid array:(NSArray<CBDescriptor *> *)array
{
    for (CBDescriptor *d in array)
    {
        if ([d.UUID isEqual:[CBUUID UUIDWithString:uuid]])
        {
            return d;
        }
    }
    return nil;
}
```

JLdevFr, 2023-11-20:
iv only use service and characteristics functions : 
flutter: /////
flutter: BluetoothService{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff2, descriptors: [BluetoothDescriptor{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, characteristicUuid: fff2, descriptorUuid: 2902, lastValue: []}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: true, notify: true, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff1, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}], includedServices: []}
flutter: BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff2, descriptors: [BluetoothDescriptor{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, characteristicUuid: fff2, descriptorUuid: 2902, lastValue: []}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: true, notify: true, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}
flutter: /////

chipweinberger, 2023-11-20:
> i think these three functions need to be this.
> 
> you can try it and let me know.

see above ^^

JLdevFr, 2023-11-20:
> > i think these three functions need to be this.
> > you can try it and let me know.
> 
> see above ^^

flutter: BluetoothService{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, isPrimary: true, characteristics: [BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff2, descriptors: [BluetoothDescriptor{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, characteristicUuid: fff2, descriptorUuid: 2902, lastValue: []}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: true, notify: true, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff1, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}], includedServices: []}
flutter: BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff2, descriptors: [BluetoothDescriptor{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, characteristicUuid: fff2, descriptorUuid: 2902, lastValue: []}], properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: true, notify: true, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}
flutter: BluetoothCharacteristic{remoteId: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, serviceUuid: fff0, secondaryServiceUuid: null, characteristicUuid: fff1, descriptors: [], properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}
[FBP-iOS] handleMethodCall: setNotifyValue
flutter: PlatformException(setNotifyValue, service not found 'fff0', null, null)

chipweinberger, 2023-11-20:
?

JLdevFr, 2023-11-20:
> - (CBService *)getServiceFromArray:(NSString *)uuid array:(NSArray<CBService *> *)array
> {
>     for (CBService *s in array)
>     {
>        NSLog(@"%@ %@", [s.UUID uuidStr], uuid); // add this [...]

Sorry, I didn't understand what you were asking.
Here are the logs:
[FBP-iOS] didConnectPeripheral
flutter: [FBP] \^[[1;30m[[ OnConnectionStateChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, connection_state: 1, disconnect_reason_code: null, disconnect_reason_string: nul
flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, success: 1, error_string: success, error_code: 0, mtu: 2
flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, success: 1, error_string: success, error_code: 0, mtu: 50
flutter: [FBP] \^[[1;30m<discoverServices>\^[[0m args: \^[[1;35mDD017D95-5AD9-E191-11A4-6A0358E5F0
[FBP-iOS] handleMethodCall: discoverServices
flutter: [FBP] \^[[1;30m<discoverServices>\^[[0m result: \^[[1;33mtr
[FBP-iOS] didDiscoverServices
[FBP-iOS] Found service: 0000FFF0-0000-1000-8000-00805F9B34FB
[FBP-iOS] didDiscoverCharacteristicsForService
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
[FBP-iOS] didDiscoverDescriptorsForCharacteristic
flutter: [FBP] \^[[1;30m[[ OnDiscoveredServices ]]\^[[0m result: \^[[1;33m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, success: 1, services: [{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, characteristics: [{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, descriptors: [{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, characteristic_uuid: 0000fff2-0000-1000-8000-00805f9b34fb, descriptor_uuid: 2902, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null}], characteristic_uuid: 0000fff2-0000-1000-8000-00805f9b34fb, properties: {notify: 1, indicate_encryption_required: 0, write: 1, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null}, {remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, descriptors: [], characteristic_uuid: 0000fff1-0000-1000-8000-00805f9b34fb, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 1, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, secondary_service_uuid: null}], is_primary: true, service_uuid: 0000fff0-0000-1000-8000-00805f9b34fb, included_services: []}], error_string: success, error_code:
flutter: [FBP] \^[[1;30m<readCharacteristic>\^[[0m args: \^[[1;35m{remote_id: DD017D95-5AD9-E191-11A4-6A0358E5F0C7, service_uuid: fff0, secondary_service_uuid: null, characteristic_uuid: fff
[FBP-iOS] handleMethodCall: readCharacteristic
0000fff0-0000-1000-8000-00805f9b34fb fff0

chipweinberger, 2023-11-20:
thanks.

> 0000fff0-0000-1000-8000-00805f9b34fb fff0

yes this log confirms the issue, the [fix](https://github.com/boskokg/flutter_blue_plus/issues/693#issuecomment-1818643009) here should fix it

JLdevFr, 2023-11-20:
> - (CBService *)getServiceFromArray:(NSString *)uuid array:(NSArray<CBService *> *)array
> {
>     for (CBService *s in array)
>     {
>         if ([s.UUID isEqual:[CBUUID UUIDWithString:uuid]]) [...]

Indeed, this has resolved my issue. Thank you very much for your prompt responses!

irfanismaya20, 2023-11-20:
which version is the most stable and most downloaded flutter blue plus? Please let me know

closed 2023-11-20 by JLdevFr

chipweinberger, 2023-11-20:
current version

chipweinberger, 2023-11-20:
fixed in 1.29.5