chipweinberger/flutter_blue_plus #696: [Bug]: discoverServices times out after 15 seconds in FBP versions 1.25.0 and up

danielle-h, 2023-11-22:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.29.5

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

12 SKQ1.211019.001

### Bluetooth Module

Xiaomi POCO X3 NFC phone

### What is your problem?

discoverServices times out after 15 seconds in version 1.25.0 and up. It works fine in version 1.20.0.

### Logs

```shell
Logs for 1.20.0:

D/[FBP-Android](29871): [FBP] onMethodCall: startScan
D/[FBP-Android](29871): [FBP] onMethodCall: stopScan
D/[FBP-Android](29871): [FBP] onMethodCall: connect
D/[FBP-Android](29871): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](29871): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](29871): [FBP] onMtuChanged: mtu: 247 status: 0
D/[FBP-Android](29871): [FBP] onMethodCall: discoverServices
D/[FBP-Android](29871): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](29871): [FBP] onServicesDiscovered: count: 5 status: 0
D/[FBP-Android](29871): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](29871): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](29871): [FBP] onCharacteristicWrite: uuid: 0fdab68b-115c-4142-ac71-6631365fc5fb status: 0
D/[FBP-Android](29871): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](29871): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](29871): [FBP] onCharacteristicWrite: uuid: 0fdab68b-115c-4142-ac71-6631365fc5fb status: 0
D/[FBP-Android](29871): [FBP] onMethodCall: disconnect
D/[FBP-Android](29871): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/[FBP-Android](29871): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](29871): [FBP] onMethodCall: disconnect
D/[FBP-Android](29871): [FBP] already disconnected
```

Logs for 1.25.0
```
D/[FBP-Android](30690): [FBP] onMethodCall: startScan
D/[FBP-Android](30690): [FBP] onMethodCall: stopScan
D/[FBP-Android](30690): [FBP] onMethodCall: connect
D/[FBP-Android](30690): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](30690): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](30690): [FBP] onMtuChanged: mtu: 247 status: 0
D/[FBP-Android](30690): [FBP] onMethodCall: requestMtu
D/[FBP-Android](30690): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](30690): [FBP] onMethodCall: discoverServices
D/[FBP-Android](30690): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](30690): [FBP] onMtuChanged: mtu: 247 status: 0
I/flutter (30690): Error in connect [FBP]: FlutterBluePlusException | discoverServices | fbp-code: 1 | Timed out after 15s
D/[FBP-Android](30690): [FBP] onMethodCall: disconnect
```
```

chipweinberger, 2023-11-22:
probably just a coincidence.

i'm not aware of any changes that would affect this. 

if you have more details to confirm it is a fbp problem, let me know.

Wackymax, 2023-11-24:
I am seeing the same. Can't connect to Android devices event. I keep on getting these Exceptions:
FlutterBluePlusException | requestMtu | fbp-code: 1 | Timed out after 15s

chipweinberger, 2023-11-24:
> if you have more details to confirm it is a fbp problem, let me know. for example, the specific fbp version that causes the issue. Are you sure it does not happen in 1.24.x?

bump

Wackymax, 2023-11-27:
I think I found something here. The connection fails if the manufacturer data service is not set as a primary service.

danielle-h, 2023-11-27:
> probably just a coincidence.
> 
> i'm not aware of any changes that would affect this.
> 
> if you have more details to confirm it is a fbp problem, let me know. for example, the specific fbp version that causes the issue. Are you sure it does not happen in 1.24.x?
> 
> you could test using the example app.

@chipweinberger It doesn't work in 1.24.x either. I didn't check all the releases between 1.20.0 and the current one... I can check more over the next week and get back to you. 

> I think I found something here. The connection fails if the manufacturer data service is not set as a primary service.

@Wackymax Thank you! I'm not sure what you mean by "manufacturer data service". However, in my device all the services are defined as primary services. 
I have: 
Device information
Generic attribute
Generic access
Custom service
Custom service

Any help would be most appreciated.

danielle-h, 2023-11-27:
> It's the service that advertises the manufacturer name and model number. Service '180a' and characteristics '2a29' and '2a24'

My device has service 180a (primary) and characteristic 2a29. It doesn't have 2a24. Could this be connected?

Wackymax, 2023-11-27:
Possibly? Try adding it and check? It's the model number being set on that one.

danielle-h, 2023-11-27:
I can't change the device firmware, only the app that talks to it.

But I'll try and narrow down the fbp version that the bug appears, it worked just fine before.

chipweinberger, 2023-11-27:
yes, knowing the specific fbp version will be very useful

choupilaopo, 2023-11-29:
我在小米手机上也遇到了同样的问题 其他手机都是正常的：我的版本是1.29.5

danielle-h, 2023-11-29:
I just saw what @choupilaopo said above. It works fine with a Samsung Galaxy phone, didn't work with Xiaomi Pocophone.
I'll get back to you on precise FBP version

manhdatbn93, 2023-11-30:
I have the same issue. It works fine on Android 10, 8, or 6 but fails on Android 13 (FBP version 1.26.9 ). I do not have Android 13 on my side, so I cannot catch the log. Do you have any ideas to resolve this? 
Firstly, I think it comes from FW on the Bluetooth device where I am programming. But I am not sure about that. I used the BlueLight App on my iPhone 13 and am still discovering all the services. Hence, I thought maybe something in Android 13 would let my device not be compatible with it. Just thought.

chipweinberger, 2023-12-01:
have you tried increasing the timeout? try 120 seconds.

chipweinberger, 2023-12-04:
@Wackymax you could set `mtu` to bull in `connect` arguments.

but this is not a good thing to see. your device may have issues.

manhdatbn93, 2023-12-04:
> have you tried increasing the timeout? try 120 seconds.
> 
> also, please try using the example app

Hi @chipweinberger, I tried the example app and still face this issue. Increasing the timeout is not working. The device will be disconnected automatically after calling getService.

chipweinberger, 2023-12-04:
your device is crashing or bugging out.

chipweinberger, 2023-12-04:
To get this issue fixed, I'll need someone to find the specific fbp version that first caused the timeout issue.

closed 2023-12-04 by chipweinberger

manhdatbn93, 2023-12-19:
`I/flutter (27661): [FBP] <discoverServices> args: 30:E8:E4:D0:20:83
D/[FBP-Android](27661): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(27661): discoverServices() - device: 30E8E4_3
D/BluetoothGatt(27661): onSearchComplete() = Device=30E8E4_3 Status=0
D/[FBP-Android](27661): [FBP] onServicesDiscovered: count: 5 status: 0
I/flutter (27661): [FBP] <discoverServices> result: true
I/flutter (27661): [FBP] [[ OnDiscoveredServices ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: 30:E8:E4:D0:20:83, error_code: 0, services: [{included_services: [], characteristics: [{descriptors: [], service_uuid: 1800, remote_id: 30:E8:E4:D0:20:83, characteristic_uuid: 2a00, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1800, remote_id: 30:E8:E4:D0:20:83, characteristic_uuid: 2a01, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1800, remote_id: 30:E8:E4:D0:20:83, characteristic_uuid: 2a04, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenti
I/flutter (27661): [FBP] <setNotifyValue> args: {remote_id: 30:E8:E4:D0:20:83, service_uuid: 1801, secondary_service_uuid: null, characteristic_uuid: 2a05, force_indications: false, enable: true}
D/[FBP-Android](27661): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(27661): setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
I/flutter (27661): [FBP] <setNotifyValue> result: true
D/BluetoothGatt(27661): onClientConnectionState() - status=0 clientIf=6 device=30E8E4_3
D/[FBP-Android](27661): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(27661): close()
D/BluetoothGatt(27661): unregisterApp() - mClientIf=6
I/flutter (27661): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 30:E8:E4:D0:20:83, connection_state: 0}`

chipweinberger, 2023-12-19:
where in your logs is the timeout?  It looks like a different issue....

manhdatbn93, 2023-12-19:
Somehow I see the timeout logs on application but cannot catch it in debug mode.

> where in your logs is the timeout? It looks like a different issue....
> 
Even if I comment out these line in FBP, it still display on log.
> It seems like some problem with your device
> 
> > I don't know why the setNotifyValue is called
> 
> ```
> // in order to match iOS behavior on all platforms,
> // we always listen to the Services Changed characteristic if it exists.
> ```

chipweinberger, 2023-12-20:
in 1.30.8 I added a new argument to `discoverServices` 

you can use it to skip the automatic  `setNotifyValue`

manhdatbn93, 2023-12-20:
> in 1.30.8 I added a new argument to `discoverServices`
> 
> you can use it to skip the automatic `setNotifyValue` for the Services Changed Characteristic

Hi @chipweinberger, the issue come from IoT device. It seems that something is conflicted inside when both BLE and Zigbee protocol are working at the same time. I remove the Zigbee part in my code and now FBP is working perfectly.

dlewis2017, 2024-10-15:
I think I'm facing the same issue. First it manifested as if setNotify was erroring out, then write would time out, and now after updating to the latest version of FBP I'm seeing it looks like the discoverServices is timing out, I've even tried setting it to . Thoughts on how to address this issue?

This may be the line that errors but i'm not sure because it either errors out or hangs: `await FlutterBluePlus._invokeMethod('discoverServices', remoteId.str);`

### Flutter Blue Plus Version
flutter_blue_plus: 1.33.4

### Flutter Env
```
[✓] Flutter (Channel beta, 3.26.0-0.1.pre, on macOS 15.0.1 24A348 darwin-arm64, locale en-US)
    • Flutter version 3.26.0-0.1.pre on channel beta at /Users/mymac/Source/SDK/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision ee624bc4fd (5 weeks ago), 2024-09-10 17:41:06 -0500
    • Engine revision 059e4e6d8f
    • Dart version 3.6.0 (build 3.6.0-216.1.beta)
    • DevTools version 2.39.0

[✓] Android toolchain - develop for Android devices (Android SDK version 32.0.0)
    • Android SDK at /Users/mymac/Library/Android/sdk
    • Platform android-32, build-tools 32.0.0
    • Java binary at: /usr/bin/java
    • Java version Java(TM) SE Runtime Environment (build 19.0.2+7-44)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 16.0)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Build 16A242d
    • CocoaPods version 1.15.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[!] Android Studio (not installed)
    • Android Studio not found; download from https://developer.android.com/studio/index.html
      (or visit https://flutter.dev/to/macos-android-setup for detailed instructions).

[✓] VS Code (version 1.94.2)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.98.0

[✓] Connected device (4 available)
    •My iPad (mobile)   • 00008403-0405528E41C2035E • ios            • iOS 17.6.1 21G93
    • macOS (desktop)                 • macos                     • darwin-arm64   • macOS 15.0.1 24A348 darwin-arm64
    • Mac Designed for iPad (desktop) • mac-designed-for-ipad     • darwin         • macOS 15.0.1 24A348 darwin-arm64
    • Chrome (web)                    • chrome                    • web-javascript • Google Chrome 129.0.6668.101

[✓] Network resources
    • All expected network resources are available.

! Doctor found issues in 1 category.
```
### Device
iPad iOS 17.6.1

### Logs
Most of the time it does this:
```
flutter: [FBP] <connect> args: {remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, auto_connect: 1}
flutter: [FBP] <connect> result: true
flutter: [FBP] [[ OnConnectionStateChanged ]] result: {remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, connection_state: 1, disconnect_reason_code: null, disconnect_reason_string: null}
flutter: [FBP] [[ OnMtuChanged ]] result: {remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, success: 1, error_string: success, error_code: 0, mtu: 507}
flutter: [FBP] [[ OnConnectionStateChanged ]] result: {remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, connection_state: 0, disconnect_reason_code: 7, disconnect_reason_string: The specified device has disconnected from us.}
```

if it makes it through it will look something like this (where it will error out on write):
```
flutter: [FBP] <setNotifyValue> args: {remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, service_uuid: cc961d00-3d30-4115-90a4-1e40e9e27895, secondary_service_uuid: null, characteristic_uuid: cc961d02-3d30-4115-90a4-1e40e9e27895, force_indications: false, enable: true}
flutter: [FBP] <setNotifyValue> result: true
flutter: [FBP] [[ OnDescriptorWritten ]] result: {error_string: success, remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, secondary_service_uuid: null, characteristic_uuid: cc961d02-3d30-4115-90a4-1e40e9e27895, success: 1, value: 01000000, error_code: 0, service_uuid: cc961d00-3d30-4115-90a4-1e40e9e27895, descriptor_uuid: 2902}
flutter: [FBP] <writeCharacteristic> args: {remote_id: F779B915-7BFB-EE7A-EBAA-F94D1AAE0A0B, service_uuid: cc961d00-3d30-4115-90a4-1e40e9e27895, secondary_service_uuid: null, characteristic_uuid: cc961d01-3d30-4115-90a4-1e40e9e27895, write_type: 0, allow_long_write: 0, value: 00205766e62d376191e1c50d3cc5b8d29d95c072d688b460caeae89c0da9bb22cd53}
[... 38 lines cut ...]
    }
    try {
      await connectingDevice.device.connect(timeout: const Duration(seconds: 10), mtu: null, autoConnect: true);
    } catch (e) {
      add(BluetoothError(errorMessage: 'Failed to connect to device', errorBody: e.toString()));
      return;
    }
  }
  ```
  
  ```
  /// Handle device connection changes to THIS device and emit to overall bloc state
  Future<void> handleDeviceConnectionChange(
    BluetoothConnectionState connectionChange,
  ) async {
    final status = device?.status;
    switch (connectionChange) {
      case BluetoothConnectionState.connected:
        try {
          List<BluetoothService> services =
              await myDevice?.device.discoverServices(timeout: 120) ?? [];
          logger.d('services found: ${services.map((service) => service).toList()}');
          # Error seems to occur here
          ...
```

chipweinberger, 2024-10-15:
@dlewis2017 its not the same issue, because this issue was fixed

it looks like your peripheral is not happy. 

you probably have some bugs in its firmware.

dlewis2017, 2024-10-15:
@chipweinberger okay, thanks for the quick reply, i'll check it out