chipweinberger/flutter_blue_plus #699: [Help]: I can't detect my Bluetooth device anymore

Mehdi-Mouilbeau, 2023-11-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.22.0

### Flutter Version

3.13.8

### What OS?

Android

### OS Version

Android 10 /12 and 14

### Bluetooth Module

Honor 9X / Tab Lenovo / Pixel 7

### What is your problem?

Hello, 

For a week now I can't detect my Bluetooth device. The detection was working fine and then nothing even after several attempts. I tried from another environment, same thing. It will find my device once and then nothing.  

### Logs

```shell
D/[FBP-Android](18683): [FBP] onMethodCall: startScan
I/BluetoothAdapter(18683): isLeEnabled(): ON
D/BluetoothLeScanner(18683): onScannerRegistered() - status=0 scannerId=15 mScannerId=0
W/Binder  (18683): Caught a RuntimeException from the binder stub implementation.
W/Binder  (18683): java.lang.SecurityException: Need BLUETOOTH_PRIVILEGED permission: Neither user 10143 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
W/Binder  (18683): 	at android.os.Parcel.createException(Parcel.java:2091)
W/Binder  (18683): 	at android.os.Parcel.readException(Parcel.java:2059)
W/Binder  (18683): 	at android.os.Parcel.readException(Parcel.java:2007)
W/Binder  (18683): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:1526)
W/Binder  (18683): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onScannerRegistered(BluetoothLeScanner.java:609)
W/Binder  (18683): 	at android.bluetooth.le.IScannerCallback$Stub.onTransact(IScannerCallback.java:115)
W/Binder  (18683): 	at android.os.Binder.execTransactInternal(Binder.java:1028)
W/Binder  (18683): 	at android.os.Binder.execTransact(Binder.java:1001)
D/[FBP-Android](18683): [FBP] onMethodCall: stopScan
I/BluetoothAdapter(18683): isLeEnabled(): ON
```

chipweinberger, 2023-11-24:
the logs seem pretty clear.

have you set permissions?

also it appears you are calling stopScan too quickly?

Mehdi-Mouilbeau, 2023-11-26:
here's my AndroidManifest for permissions : 

  
  <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  
  <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
  
  <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
  
  That's how the problem arose. I don't think I changed anything.
I even went back to an older commit or another branch to test and the problem is still there. 
I thought it was an environment problem, so I tested on windows and apple, same problem. 
I haven't changed the way I call stopscan since it worked like that. I can try.

chipweinberger, 2023-11-26:
it says you need BLUETOOTH_PRIVILEGED

Mehdi-Mouilbeau, 2023-11-26:
I've tried it with and without, and nothing has changed.

chipweinberger, 2023-11-26:
lmk if you figure it out

Mehdi-Mouilbeau, 2023-11-27:
Hello, 
We were able to target the problem by removing the withServices parameter from the startscan. There seems to be an incompatibility with Android when trying to access the Guid (hence the bluetooth privileged permission). 
Fyi, result is the same with or wihtout this permission in the manifest.
On the first scan it finds several Bluetooth devices, but on the next ones it finds none.

chipweinberger, 2023-11-27:
"**Restricted Permission:** The BLUETOOTH_PRIVILEGED permission is a protected-level permission. This means it's not available to all apps. Typically, only system apps or apps signed with the system's signing key can obtain this permission. Regular third-party apps cannot use this permission due to security reasons."

yes seems like you're doing something that is not allowed.

chipweinberger, 2023-11-27:
what service are you trying to scan for?

chipweinberger, 2023-11-30:
@cl-lumoshelmet which android version?

cl-lumoshelmet, 2023-11-30:
I'm running on a Huawei Mate 30 (EMUI 12), Android 10

chipweinberger, 2023-11-30:
what service are you scanning for

chipweinberger, 2023-12-03:
According to this [issue](https://github.com/pauldemarco/flutter_blue/issues/861), your bluetooth is not enabled.

Update to 1.27.6 or later. I added checks for bluetooth being off.

closed 2023-12-03 by chipweinberger