chipweinberger/flutter_blue_plus #700: [Bug]: Can't detect any devices with old android phones

Jbz797, 2023-11-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.27

### Flutter Version

2.10.5

### What OS?

Android

### OS Version

8.0.0

### Bluetooth Module

...

### What is your problem?

Since version v1.27 i can't detect any device on my old Android phone (Android 8.0.0).

### Logs

```shell
E/[FBP-Android]( 8669): [FBP] onScanFailed: SCAN_FAILED_OUT_OF_HARDWARE_RESOURCES
```

chipweinberger, 2023-11-24:
try restarting phone

irfanismaya20, 2023-11-25:
same issue can't read device info since version version ^1.27 in iOS, status is connected but can not read data state device information

irfanismaya20, 2023-11-25:
StreamSubscription<List<int>>? _deviceInfoSubs = notifyChar?.onValueReceived.listen((data) {

});

i cannot read or get data in version ^1.27 in iOS but in under version 1.27 like (1.20, 1.21, 1.26) i can get data

Jbz797, 2023-11-25:
> try restarting phone

It worked when I restarted the phone. Thank you.
What's strange is that I've never had this happen with previous versions of the library (3 years of use ^^).

closed 2023-11-25 by chipweinberger