chipweinberger/flutter_blue_plus #705: How to discover services of paired devices

AdemYavuzCelik, 2023-11-27:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.29.6

### Flutter Version

3.13.5

### What OS?

Android

### OS Version

13/14

### Bluetooth Module

esp32c3

### What is your problem?

If the device is not a paired device, connection, service discovery, etc. work very well but device is paired, my app cant discover services. Another problem, it's rare but it does happen, Sometimes it shows autoconnect true  in logs but autoconnect true or just connect anywhere in the application. Im using only autoconnect(false).

### Logs

```shell
D/[FBP-Android]( 5417): [FBP] onMethodCall: stopScan
D/BluetoothAdapter( 5417): isLeEnabled(): ON
I/flutter ( 5417): FINE    | DesiNavigator          | navigating to changewifiscreen
D/[FBP-Android]( 5417): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 5417): isLeEnabled(): ON
D/BluetoothLeScanner( 5417): onScannerRegistered() - status=0 scannerId=19 mScannerId=0
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Instance of 'BluetoothDeviceCubit' Created
I/flutter ( 5417): FINE    | ChangeSettingsCubit    | Register called
I/flutter ( 5417): FINE    | ChangeSettingsCubit    | 7C:DF:A1:BA:2D:BA
I/flutter ( 5417): FINE    | ChangeSettingsCubit    | Device is registered
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Connection State: BluetoothConnectionState.disconnected
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Connected Devices: []
I/flutter ( 5417): WARNING | ChangeSettingsCubit    | Device Disconnect
D/[FBP-Android]( 5417): [FBP] onMethodCall: getAdapterState
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Connected Devices: []
D/[FBP-Android]( 5417): [FBP] onMethodCall: connect
D/CompatibilityChangeReporter( 5417): Compat change id reported: 265103382; UID 10965; state: DISABLED
D/BluetoothGatt( 5417): connect() - device: 7C:DF:A1:BA:2D:BA, auto: false
D/BluetoothGatt( 5417): registerApp()
D/BluetoothGatt( 5417): registerApp() - UUID=1631e2a2-10a6-4768-af82-7efd9c0c6305
D/BluetoothGatt( 5417): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt( 5417): onClientConnectionState() - status=0 clientIf=6 device=7C:DF:A1:BA:2D:BA
D/[FBP-Android]( 5417): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Connection State: BluetoothConnectionState.connected
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Discovering Services
D/[FBP-Android]( 5417): [FBP] onMethodCall: connect
D/[FBP-Android]( 5417): [FBP] already connected
D/[FBP-Android]( 5417): [FBP] onMethodCall: discoverServices
D/BluetoothGatt( 5417): discoverServices() - device: 7C:DF:A1:BA:2D:BA
D/[FBP-Android]( 5417): [FBP] onMethodCall: getAdapterState
D/BluetoothGatt( 5417): onSearchComplete() = Device=7C:DF:A1:BA:2D:BA Status=0
D/[FBP-Android]( 5417): [FBP] onServicesDiscovered: count: 4 status: 0
E/ScanRecord( 5417): unable to parse scan record: [30, 5, 18, -94, 77, 46, -2, 20, 72, -114, -109, -46, 23, 60, -3, 2, 48, -121, 126, 7, -40, -95, 110, 0, 27, 40, 64, 0, 0, 0, 0]
D/[FBP-Android]( 5417): [FBP] onMethodCall: requestMtu
D/BluetoothGatt( 5417): configureMTU() - device: 7C:DF:A1:BA:2D:BA mtu: 512
[... 76 lines cut ...]
D/BluetoothGatt( 5417): close()
D/BluetoothGatt( 5417): unregisterApp() - mClientIf=6
D/[FBP-Android]( 5417): [FBP] onMethodCall: setNotifyValue
I/flutter ( 5417): FINE    | BluetoothDeviceCubit   | Connection State: BluetoothConnectionState.disconnected
I/flutter ( 5417): WARNING | ChangeSettingsCubit    | Device Disconnect
I/flutter ( 5417): SEVERE  | DesiErrorHandler       | caught error
I/flutter ( 5417): --------------------------------
I/flutter ( 5417): ERROR
I/flutter ( 5417): PlatformException(setNotifyValue, device is disconnected, null, null)
I/flutter ( 5417): --------------------------------
I/flutter ( 5417): STACK TRACE
I/flutter ( 5417): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
I/flutter ( 5417): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
I/flutter ( 5417): <asynchronous suspension>
I/flutter ( 5417): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:499:13)
I/flutter ( 5417): <asynchronous suspension>
I/flutter ( 5417): #3      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:269:22)
I/flutter ( 5417): <asynchronous suspension>
I/flutter ( 5417): #4      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:251:9)
I/flutter ( 5417): <asynchronous suspension>
I/flutter ( 5417): #5      BluetoothDeviceCubit.discoverServices (package:desi/core/services/bluetooth_device_cubit/bluetooth_device_cubit.dart:122:42)
I/flutter ( 5417): <asynchronous suspension>
I/flutter ( 5417): --------------------------------
I/flutter ( 5417): INFO    | DesiErrorHandler       | not reporting error due to missing consent from the user
```

closed 2023-11-27 by chipweinberger

chipweinberger, 2023-11-27:
in 1.5.2 i'm pretty sure most errors were just ignored 

feel free to catch and ignore the errors if it still works

AdemYavuzCelik, 2023-11-27:
This failure not happen on Xiaomi device, but it does on Google Pixel 6. Can we still say that the IoT device is buggy?

chipweinberger, 2023-11-27:
from the logs it seems you're device might not like "requestMtu"

try setting mtu to null 

i.e `device.connect(mtu: null);`

AdemYavuzCelik, 2023-11-27:
We have to set mtu 512.  i cant set null :/