chipweinberger/flutter_blue_plus #706: [Bug]: Running in macOS, getting BluetoothAdapterState.unavailable every time and my bluetooth is enable.

karansingla007, 2023-11-28:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.29.6

### Flutter Version

channel stable 3.13.9

### What OS?

macOS

### OS Version

macOS 14.1.1

### Bluetooth Module

BLE

### What is your problem?

```
FlutterBluePlus.adapterState.listen((state) {
      _adapterState = state;
      setState(() {});
    });
```

Getting BluetoothAdapterState.unavailable everytime. I am running in MacOs.

### Logs

```shell
2023-11-28 11:31:30.725 aavaa_flutter[4496:1446362] [FBP-iOS] handleMethodCall: flutterHotRestart
2023-11-28 11:31:30.725 aavaa_flutter[4496:1446362] [FBP-iOS] disconnectAllDevices(flutterHotRestart)
2023-11-28 11:31:30.725 aavaa_flutter[4496:1446362] [FBP-iOS] connectedPeripherals: 0
flutter: [FBP] <setLogLevel> args: 5
2023-11-28 11:31:30.728 aavaa_flutter[4496:1446362] [FBP-iOS] handleMethodCall: setLogLevel
flutter: [FBP] <setLogLevel> result: true
flutter: [FBP] <getAdapterState> args: null
2023-11-28 11:31:31.743 aavaa_flutter[4496:1446362] [FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] <getAdapterState> result: {adapter_state: 1}
```

chipweinberger, 2023-11-28:
never seen this issue before.

it's normal for it to be unavailable breifly.

karansingla007, 2023-11-28:
@chipweinberger  how can i solve this issue. I couldn't able to scan device.

chipweinberger, 2023-11-28:
perhaps you need to call some other FBP method to get it to initialize?

karansingla007, 2023-11-28:
Which method, would you like to suggest?

chipweinberger, 2023-11-28:
nevermind, i'm wrong. setLogLevel is good enough. 

this seems like a macOS issue, not FBP issue

try restarting computer

karansingla007, 2023-11-28:
i am already doing this.

```
FlutterBluePlus.setLogLevel(LogLevel.verbose, color: true);
```

chipweinberger, 2023-11-28:
try restarting computer

karansingla007, 2023-11-28:
Just did, Still the problem is same.

chipweinberger, 2023-12-04:
make sure you've added permissions. see readme

chipweinberger, 2023-12-05:
@maham the problem is that `FlutterBluePlus.adapterState` gets stuck on `unavailable`

@karansingla007 any updates?

thiruarasu16, 2023-12-11:
I am also facing same unavailable problem.

chipweinberger, 2023-12-11:
Thanks @thiruarasu16 , I've updated the README & Common Problems.

closed 2023-12-11 by chipweinberger