chipweinberger/flutter_blue_plus #708: [Bug]: Bug crash app for android

tqdaoinfo108, 2023-11-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.29.7

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

android

### Bluetooth Module

Esp32

### What is your problem?

[image]
Maybe it's casting the wrong type

### Logs

```shell
E/AndroidRuntime(24857): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$3.onScanResult(FlutterBluePlusPlugin.java:1882)
E/AndroidRuntime(24857): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:572)
E/AndroidRuntime(24857): 	at android.os.Handler.handleCallback(Handler.java:942)
E/AndroidRuntime(24857): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/AndroidRuntime(24857): 	at android.os.Looper.loopOnce(Looper.java:210)
E/AndroidRuntime(24857): 	at android.os.Looper.loop(Looper.java:299)
E/AndroidRuntime(24857): 	at android.app.ActivityThread.main(ActivityThread.java:8261)
E/AndroidRuntime(24857): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(24857): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:559)
E/AndroidRuntime(24857): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:954)
W/OOMEventManagerFK(24857): Failed to mkdir /data/miuilog/stability/memleak/heapdump/
E/MQSEventManagerDelegate(24857): failed to get MQSService.
I/Process (24857): Process is going to kill itself!
I/Process (24857): java.lang.Exception
I/Process (24857): 	at android.os.Process.killProcess(Process.java:1344)
I/Process (24857): 	at com.android.internal.os.RuntimeInit$KillApplicationHandler.uncaughtException(RuntimeInit.java:181)
I/Process (24857): 	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1073)
I/Process (24857): 	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1068)
I/Process (24857): 	at java.lang.Thread.dispatchUncaughtException(Thread.java:2306)
I/Process (24857): Sending signal. PID: 24857 SIG: 9
Lost connection to device.
```

mabin, 2023-11-29:
same problem : 
E/AndroidRuntime(30132): java.lang.ClassCastException: java.lang.Boolean cannot be cast to java.lang.Integer
E/AndroidRuntime(30132): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$3.onScanResult(FlutterBluePlusPlugin.java:1882)
E/AndroidRuntime(30132): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:646)
E/AndroidRuntime(30132): 	at android.os.Handler.handleCallback(Handler.java:900)
E/AndroidRuntime(30132): 	at android.os.Handler.dispatchMessage(Handler.java:103)
E/AndroidRuntime(30132): 	at android.os.Looper.loop(Looper.java:219)
E/AndroidRuntime(30132): 	at android.app.ActivityThread.main(ActivityThread.java:8673)
E/AndroidRuntime(30132): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(30132): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
E/AndroidRuntime(30132): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109)

tqdaoinfo108, 2023-11-29:
@mabin  In the meantime, you can use it
```
flutter_blue_plus:
    git: 
      url: https://github.com/tqdaoinfo108/flutter_blue_plus_tq.git
      ref: master
```

kashjack, 2023-11-29:
thank you, it works for me！

chipweinberger, 2023-11-29:
fixed 1.29.8

closed 2023-11-29 by chipweinberger

kashjack, 2023-11-29:
👍🏻

michael-kim-korea, 2023-11-30:
I searched all day but found the answer here.
👍🏻

michael-kim-korea, 2023-11-30:
> fixed 1.29.8
I searched all day but found the answer here.
👍🏻