chipweinberger/flutter_blue_plus #711: [Bug]:

Bilal1025, 2023-11-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.29.6

### Flutter Version

3.16.1

### What OS?

Android

### OS Version

ONE UI 5.1, Android - 13

### Bluetooth Module

EFR32xG22 2.4 GHz 6 dBm Radio Board (BRD4182A Rev B06)

### What is your problem?

calling `setNotifyValue(true)` on characteristic returns `Subscribe Error: GATT_WRITE_NOT_PERMITTED` on `0x2A05` (Services Changed characteristic), it does have notify property enabled. I am using OTA functionality to flash firmware to hardware, but when in OTA mode `discoverServices()` crashes due to this.

Note: it works fine on kotlin/Java (using native android code)

### Logs

```shell
I/ViewRootImpl@4f7c3b4[MainActivity](10698): ViewPostIme pointer 0
I/ViewRootImpl@4f7c3b4[MainActivity](10698): ViewPostIme pointer 1
I/flutter (10698): [FBP] <setNotifyValue> args: {remote_id: 38:5B:44:BB:0C:64, service_uuid: 1801, secondary_service_uuid: null, characteristic_uuid: 2a05, force_indications: false, enable: true}
D/[FBP-Android](10698): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(10698): setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
I/flutter (10698): [FBP] <setNotifyValue> result: true
I/flutter (10698): [FBP] <getAdapterState> args: null
D/[FBP-Android](10698): [FBP] onMethodCall: getAdapterState
I/flutter (10698): [FBP] <getAdapterState> result: {adapter_state: 4}
D/[FBP-Android](10698): [FBP] onDescriptorWrite: 2902 status: 3
I/flutter (10698): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_WRITE_NOT_PERMITTED, service_uuid: 1801, success: 0, remote_id: 38:5B:44:BB:0C:64, descriptor_uuid: 2902, error_code: 3, characteristic_uuid: 2a05, value: }
```

chipweinberger, 2023-11-30:
this error is a native error

I don't think FBP can fix it.

let me know if you figure it out

chipweinberger, 2023-11-30:
btw, you should use the FBP code to detect services changed

```
  /// Services Reset Stream
  ///  - uses the GAP Services Changed characteristic (0x2A05)
  ///  - you must re-call discoverServices()
  Stream<void> get onServicesReset 
```

Bilal1025, 2023-12-01:
@chipweinberger I've checked/verified that the 2A05 support being subscribed to, however when calling `setNotifyValue(true)` on it, I am getting that particular exception thrown. Even when I call `discoverServices` it throws that exception. I see that we are checking 
```
if (c != null && (c.properties.notify || c.properties.indicate) && c.isNotifying == false) {
```

So if it doesn't support subscribing it shouldn't get called in the first place. But it gets called and crashes on this ` await c.setNotifyValue(true);` and throws that exception

chipweinberger, 2023-12-01:
what happens on ios?

chipweinberger, 2023-12-01:
"Example in Android: In an Android application, when you have a reference to a BluetoothGattDescriptor object, you can use the getPermissions() method to check its permissions. This method returns an integer, and you can use bitwise AND operations with flags like BluetoothGattDescriptor.PERMISSION_READ or BluetoothGattDescriptor.PERMISSION_WRITE to determine the permissions."

https://developer.android.com/reference/android/bluetooth/BluetoothGattDescriptor

looks like FBP needs to check these fields

chipweinberger, 2023-12-01:
yes you can still subscribe, but its the writing of the CCCD that causes the problem.

I imagine your kotlin code will have thr same issue if you update the CCCD

chipweinberger, 2023-12-02:
Thinking about it more, I'll leave the code as is. The user can catch the exception.