chipweinberger/flutter_blue_plus #712: [Bug]: Bluetooth disconnects after write

FritzMatthaeus, 2023-11-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.29.8

### Flutter Version

3.16.1

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

unknown, supports Bluetooth 5.1, A2DP, LE

### What is your problem?

I write a value (without response) to a Bluetooth Characteristic and then the bluetooth connection gets disconnected. Any help on intepreting the native logs is appreciated, thx.

### Logs

```shell
D/[FBP-Android](30981): [FBP] onScanResult
I/flutter (30981): withoutResponse true
I/flutter (30981): [FBP] <writeCharacteristic> args: {remote_id: 6C:1D:EB:04:AA:78, service_uuid: b7c529bc-970e-4879-b29f-b6d01d6d91e3, secondary_service_uuid: null, characteristic_uuid: 7a25dfc1-c147-4a86-a4a7-255654314e7c, write_type: 1, allow_long_write: 0, value: 02b3b20000100020f1d80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd8000021780000edd8000080140200b1780000a5920000edd800005d660000c9d40000a5d10000edd80000edd80000edd80000edd80000edd80000edd80000edd80000d918000025190000c1d10000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd800001d9a0000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd80000edd8000010b5054c237833b9044b13b10448aff300800123237010bda85c0020000000005c34020008b5034b1bb103490348aff3008008bd00000000ac5c00205c340200a3f5803a7047
D/[FBP-Android](30981): [FBP] onMethodCall: writeCharacteristic
4
D/[FBP-Android](30981): [FBP] onScanResult
I/flutter (30981): [FBP] <writeCharacteristic> result: true
2
D/[FBP-Android](30981): [FBP] onScanResult
I/flutter (30981): [FBP] <getAdapterState> args: null
10
D/[FBP-Android](30981): [FBP] onScanResult
D/[FBP-Android](30981): [FBP] onMethodCall: getAdapterState
I/flutter (30981): [FBP] <getAdapterState> result: {adapter_state: 4}
37
D/[FBP-Android](30981): [FBP] onScanResult
D/BluetoothAdapter(30981): onBluetoothServiceDown
D/[FBP-Android](30981): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](30981): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](30981): [FBP] calling disconnect: 6C:1D:EB:04:AA:78
D/BluetoothGatt(30981): cancelOpen() - device: 6C1DEB_8
E/BluetoothGatt(30981): android.os.DeadObjectException
E/BluetoothGatt(30981): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt(30981): 	at android.os.BinderProxy.transact(BinderProxy.java:662)
E/BluetoothGatt(30981): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.clientDisconnect(IBluetoothGatt.java:1901)
E/BluetoothGatt(30981): 	at android.bluetooth.BluetoothGatt.disconnect(BluetoothGatt.java:1182)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.disconnectAllDevices(FlutterBluePlusPlugin.java:1686)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.access$200(FlutterBluePlusPlugin.java:78)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1755)
E/BluetoothGatt(30981): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1919)
E/BluetoothGatt(30981): 	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
E/BluetoothGatt(30981): 	at android.os.Handler.handleCallback(Handler.java:942)
E/BluetoothGatt(30981): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/BluetoothGatt(30981): 	at android.os.Looper.loopOnce(Looper.java:226)
E/BluetoothGatt(30981): 	at android.os.Looper.loop(Looper.java:313)
E/BluetoothGatt(30981): 	at android.app.ActivityThread.main(ActivityThread.java:8762)
E/BluetoothGatt(30981): 	at java.lang.reflect.Method.invoke(Native Method)
E/BluetoothGatt(30981): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
E/BluetoothGatt(30981): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
D/[FBP-Android](30981): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(30981): close()
D/BluetoothAdapter(30981): onBluetoothServiceDown
D/BluetoothGatt(30981): unregisterApp() - mClientIf=8

E/BluetoothGatt(30981): android.os.DeadObjectException
E/BluetoothGatt(30981): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt(30981): 	at android.os.BinderProxy.transact(BinderProxy.java:662)
E/BluetoothGatt(30981): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.unregisterClient(IBluetoothGatt.java:1859)
E/BluetoothGatt(30981): 	at android.bluetooth.BluetoothGatt.unregisterApp(BluetoothGatt.java:1089)
E/BluetoothGatt(30981): 	at android.bluetooth.BluetoothGatt.close(BluetoothGatt.java:938)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$4.onConnectionStateChange(FlutterBluePlusPlugin.java:1987)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.disconnectAllDevices(FlutterBluePlusPlugin.java:1692)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.access$200(FlutterBluePlusPlugin.java:78)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1755)
E/BluetoothGatt(30981): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1919)
E/BluetoothGatt(30981): 	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
E/BluetoothGatt(30981): 	at android.os.Handler.handleCallback(Handler.java:942)
E/BluetoothGatt(30981): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/BluetoothGatt(30981): 	at android.os.Looper.loopOnce(Looper.java:226)
E/BluetoothGatt(30981): 	at android.os.Looper.loop(Looper.java:313)
E/BluetoothGatt(30981): 	at android.app.ActivityThread.main(ActivityThread.java:8762)
E/BluetoothGatt(30981): 	at java.lang.reflect.Method.invoke(Native Method)
E/BluetoothGatt(30981): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
E/BluetoothGatt(30981): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
D/[FBP-Android](30981): [FBP] calling close: 6C:1D:EB:04:AA:78
D/BluetoothGatt(30981): close()
D/BluetoothGatt(30981): unregisterApp() - mClientIf=8
E/BluetoothGatt(30981): android.os.DeadObjectException
E/BluetoothGatt(30981): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt(30981): 	at android.os.BinderProxy.transact(BinderProxy.java:662)
E/BluetoothGatt(30981): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.unregisterClient(IBluetoothGatt.java:1859)
E/BluetoothGatt(30981): 	at android.bluetooth.BluetoothGatt.unregisterApp(BluetoothGatt.java:1089)
E/BluetoothGatt(30981): 	at android.bluetooth.BluetoothGatt.close(BluetoothGatt.java:938)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.disconnectAllDevices(FlutterBluePlusPlugin.java:1706)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.access$200(FlutterBluePlusPlugin.java:78)
E/BluetoothGatt(30981): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1755)
E/BluetoothGatt(30981): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1919)
E/BluetoothGatt(30981): 	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
E/BluetoothGatt(30981): 	at android.os.Handler.handleCallback(Handler.java:942)
E/BluetoothGatt(30981): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/BluetoothGatt(30981): 	at android.os.Looper.loopOnce(Looper.java:226)
E/BluetoothGatt(30981): 	at android.os.Looper.loop(Looper.java:313)
E/BluetoothGatt(30981): 	at android.app.ActivityThread.main(ActivityThread.java:8762)
E/BluetoothGatt(30981): 	at java.lang.reflect.Method.invoke(Native Method)
E/BluetoothGatt(30981): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
E/BluetoothGatt(30981): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
I/flutter (30981): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 6C:1D:EB:04:AA:78, connection_state: 0}
D/[FBP-Android](30981): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](30981): [FBP] disconnectAllDevices(adapterTurnOff)
```

joyhope, 2023-11-30:
Do you have another BLE tool, write the value to your BLE device, and check whether this write causes the problem.  It may be your device's response to the unknown data.

FritzMatthaeus, 2023-11-30:
> Do you have another BLE tool, write the value to your BLE device, and check whether this write causes the problem. It may be your device's response to the unknown data.

thanks for the hint. I've found the solution. By mistake i sent a byte array with 100000 bytes in a single write request. Normally this would be blocked by FBP, but in our implementation we ignore the mtu size check on write requests. Obviously this large payload crashed the connection.

problem solved