chipweinberger/flutter_blue_plus #713: [Bug]: isNotifying Don't Change Value

ms-daniel, 2023-11-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.29.5

### Flutter Version

3.16.0

### What OS?

Android

### OS Version

Android 13 TKQ1.221114.001

### Bluetooth Module

Esp32

### What is your problem?

When subscribing to a characteristics, the value of the Subscribe button does not change and the value of the c.isNotifying variable does not change either.

### Logs

```shell
D/[FBP-Android](12226): [FBP] onCharacteristicChanged: beb5483e-36e1-4688-b7f5-ea07361b26a8
I/flutter (12226): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, success: 1, remote_id: D8:BC:38:E5:05:7E, error_code: 0, characteristic_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, value: 48656c6c6f20576f726c6420313134}
I/flutter (12226): [FBP] <setNotifyValue> args: {remote_id: D8:BC:38:E5:05:7E, service_uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, secondary_service_uuid: null, characteristic_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, force_indications: false, enable: true}
D/[FBP-Android](12226): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(12226): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8 enable: true
I/flutter (12226): [FBP] <setNotifyValue> result: true
I/flutter (12226): [FBP] <getAdapterState> args: null
D/[FBP-Android](12226): [FBP] onMethodCall: getAdapterState
I/flutter (12226): [FBP] <getAdapterState> result: {adapter_state: 4}
D/[FBP-Android](12226): [FBP] onDescriptorWrite: 2902 status: 0
I/flutter (12226): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, success: 1, remote_id: D8:BC:38:E5:05:7E, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, value: }
I/flutter (12226): [FBP] <readCharacteristic> args: {remote_id: D8:BC:38:E5:05:7E, service_uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, secondary_service_uuid: null, characteristic_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8}
D/[FBP-Android](12226): [FBP] onMethodCall: readCharacteristic
I/flutter (12226): [FBP] <readCharacteristic> result: true
I/flutter (12226): [FBP] <getAdapterState> args: null
D/[FBP-Android](12226): [FBP] onMethodCall: getAdapterState
I/flutter (12226): [FBP] <getAdapterState> result: {adapter_state: 4}
D/[FBP-Android](12226): [FBP] onCharacteristicRead: beb5483e-36e1-4688-b7f5-ea07361b26a8 status: 0
I/flutter (12226): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, success: 1, remote_id: D8:BC:38:E5:05:7E, error_code: 0, characteristic_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, value: 48656c6c6f20576f726c6420313134}
```

ms-daniel, 2023-11-30:
the setNotifyValue method works, so I created a global variable to represent isNotifying instead of using the class one. It works, but it's bad code. The isNotifying method still does not work even when sending a setNotifyValue(true).

ms-daniel, 2023-11-30:
Thanks, @chipweinberger . I'm new to flutter/dart and haven't found the android code yet. But I will look for and test your solution

chipweinberger, 2023-11-30:
fixed in 1.29.10

closed 2023-11-30 by chipweinberger

chipweinberger, 2023-11-30:
actually use 1.29.11