chipweinberger/flutter_blue_plus #714: [Bug]: disconnect does not disconnect

FritzMatthaeus, 2023-11-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.29.8

### Flutter Version

3.16.1

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

ANNA-B112

### What is your problem?

I can successfully connect my Samsung A53 to the Bluetooth Device as i can see in it's logs.  When calling disconnect via FBP, i get a success message from FBP but the connection is still alive. I do not get a disconnect message in the Bluetooth device logs and when checking the connection with the NrfConnect App i can see, that the connection is still alive. 

If i connect and disconnect to the Device via the NrfConnect App it successfully get's connected and disconnected. So this problem is related to FBP and not to the Bluetooth Device. 

### Logs

```shell
I/ViewRootImpl@70d14a0[MainActivity](27532): ViewPostIme pointer 0
I/ViewRootImpl@70d14a0[MainActivity](27532): ViewPostIme pointer 1
I/flutter (27532): [FBP] <getAdapterState> args: null
D/[FBP-Android](27532): [FBP] onMethodCall: getAdapterState
I/flutter (27532): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (27532): [FBP] <connect> args: {remote_id: 6C:1D:EB:04:AA:78, auto_connect: 0}
D/[FBP-Android](27532): [FBP] onMethodCall: connect
I/BluetoothAdapter(27532): STATE_ON
D/BluetoothGatt(27532): connect() - device: 6C1DEB_8, auto: false
D/BluetoothGatt(27532): registerApp()
D/BluetoothGatt(27532): registerApp() - UUID=37ed384e-56c3-4536-acc1-9fffeec91d73
D/BluetoothGatt(27532): onClientRegistered() - status=0 clientIf=8
W/OnBackInvokedCallback(27532): OnBackInvokedCallback is not enabled for the application.
W/OnBackInvokedCallback(27532): Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
I/flutter (27532): [FBP] <connect> result: true
I/flutter (27532): [FBP] <getAdapterState> args: null
D/[FBP-Android](27532): [FBP] onMethodCall: getAdapterState
I/flutter (27532): [FBP] <getAdapterState> result: {adapter_state: 4}
D/BluetoothGatt(27532): onClientConnectionState() - status=0 clientIf=8 device=6C1DEB_8
D/[FBP-Android](27532): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
I/flutter (27532): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 6C:1D:EB:04:AA:78, connection_state: 1}
I/flutter (27532): [FBP] <readRssi> args: 6C:1D:EB:04:AA:78
D/[FBP-Android](27532): [FBP] onMethodCall: readRssi
D/BluetoothGatt(27532): readRssi() - device: 6C1DEB_8
D/[FBP-Android](27532): [FBP] onReadRemoteRssi: rssi: -46 status: 0
I/flutter (27532): [FBP] <readRssi> result: true
I/flutter (27532): [FBP] <getAdapterState> args: null
D/[FBP-Android](27532): [FBP] onMethodCall: getAdapterState
I/flutter (27532): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -46, success: 1, remote_id: 6C:1D:EB:04:AA:78, error_code: 0}
I/flutter (27532): [FBP] <getAdapterState> result: {adapter_state: 4}
D/BluetoothGatt(27532): onConnectionUpdated() - Device=6C1DEB_8 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(27532): onConnectionUpdated() - Device=6C1DEB_8 interval=36 latency=0 timeout=500 status=0
I/ViewRootImpl@70d14a0[MainActivity](27532): ViewPostIme pointer 0
I/ViewRootImpl@70d14a0[MainActivity](27532): ViewPostIme pointer 1
I/flutter (27532): [FBP] <disconnect> args: 6C:1D:EB:04:AA:78
D/[FBP-Android](27532): [FBP] onMethodCall: disconnect
D/[FBP-Android](27532): [FBP] disconnect: canceling connection in progress
D/BluetoothGatt(27532): cancelOpen() - device: 6C1DEB_8
D/BluetoothGatt(27532): onClientConnectionState() - status=0 clientIf=8 device=6C1DEB_8
I/flutter (27532): [FBP] <disconnect> result: true
I/flutter (27532): [FBP] <getAdapterState> args: null
D/[FBP-Android](27532): [FBP] onMethodCall: getAdapterState
I/flutter (27532): [FBP] <getAdapterState> result: {adapter_state: 4}
D/[FBP-Android](27532): [FBP] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(27532): close()
D/BluetoothGatt(27532): unregisterApp() - mClientIf=8
I/flutter (27532): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 6C:1D:EB:04:AA:78, connection_state: 0}
```

chipweinberger, 2023-11-30:
>  [FBP] disconnect: canceling connection in progress

this is also weird because your logs show the connection already finished. This log is only supposed to happen if establishing the connection is still in progress.

perhaps you can try to debug that issue ^

chipweinberger, 2023-11-30:
@FritzMatthaeus , `disconnect` only disconnects *our app*. if another app is connected it will not do anything AFAIK.

FritzMatthaeus, 2023-11-30:
> It must be possible somehow, even if it hasn't been implemented yet. 
> 

I've tried BluetoothGatt.close() but this did not work.

FritzMatthaeus, 2023-11-30:
> >  [FBP] disconnect: canceling connection in progress
> 
> 
> 
> this is also weird because your logs show the connection already finished. This log is only supposed to happen if establishing the connection is still in progress.
> 

As far as i understood, this log will always show up on disconnect, so it does not mesn that the connection is establishing.

chipweinberger, 2023-11-30:
no i don't think that is possible.

your app is not allowed to control other apps.

FritzMatthaeus, 2023-12-01:
> no i don't think that is possible.
> 
> your app is not allowed to control other apps connections.
> 

But i can disconnect a connection established in a FBP Demo App with the NrfConnect App. So obviously it is possible somehow. 

Another explanation might be, that unintentionally there is more than one gatt instance created by FBP so disconnecting would obly disconnect the one instance but there is still another one alive. But even in this scenario i am able to disconnect via nRFConnect App.

chipweinberger, 2023-12-01:
thanks testing it. 

Im not sure what the issue could be then.

If you figure out more details let me know.

chipweinberger, 2023-12-05:
by the way, `disconnect` works on Android 13 for me.

When I press disconnect in the example app, it disconnects in the device firmware.

FritzMatthaeus, 2023-12-05:
Hi,

yes, it works fine. The reason for the problem has been multiple native GATT-Instances with the same Device. As you can only disconnect one via the App, the Connection stays alive as further Instances are connected. I assume, this happened as i used the hot reload feature while beeing connected and this might have created new Instances with the same device. But i cannot exclude that this might be a bug in the library. This might need further inspection

closed 2023-12-05 by FritzMatthaeus

chipweinberger, 2023-12-05:
if you find reliable repro steps, please open a new issue.

FritzMatthaeus, 2023-12-05:
It can be finally closed as i could reproduce it. I have the nRFConnect App installed for testing purposes. When establishing the connection via my FBP-App i get asked in a notification if i wanted to debug the connection with the nRFConnect App. If i click "yes" (which i did, as i did not pay attention to it), it will establish a GATT-Instance in nRF-Connect so disconnecting in my FBP-App will not fully disconnect the device. So this is a special case when debugging with nRF-Connect