chipweinberger/flutter_blue_plus #715: [Bug]: Connect Error

baoxu, 2023-12-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.30.2

### Flutter Version

3.16.2

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

OPPO A2 Pro

### What is your problem?

Connect Error

### Logs

```shell
I/flutter (22902): │ 💡 Connect Error:, PlatformException(androidException, java.lang.ClassCastException: java.util.HashMap cannot be cast to java.lang.String, java.lang.ClassCastException: java.util.HashMap cannot be cast to java.lang.String
I/flutter (22902): │ 💡 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$5$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:664)
I/flutter (22902): │ 💡 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda6.op(Unknown Source:6)
I/flutter (22902): │ 💡 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissions(FlutterBluePlusPlugin.java:1493)
I/flutter (22902): │ 💡 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:650)
I/flutter (22902): │ 💡 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:267)
I/flutter (22902): │ 💡 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter (22902): │ 💡 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
I/flutter (22902): │ 💡 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter (22902): │ 💡 	at android.os.Handler.handleCallback(Handler.java:942)
I/flutter (22902): │ 💡 	at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter (22902): │ 💡 	at android.os.Looper.loopOnce(Looper.java:240)
I/flutter (22902): │ 💡 	at android.os.Looper.loop(Looper.java:351)
I/flutter (22902): │ 💡 	at android.app.ActivityThread.main(ActivityThread.java:8423)
I/flutter (22902): │ 💡 	at java.lang.reflect.Method.invoke(Native Method)
I/flutter (22902): │ 💡 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:568)
I/flutter (22902): │ 💡 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1041)
I/flutter (22902): │ 💡 , null)
```

utamori, 2023-12-05:
same problem

nagosoo, 2023-12-05:
same problem

thangbqgreenifyvn, 2023-12-05:
same problem

chipweinberger, 2023-12-05:
fixed 1.30.3

closed 2023-12-05 by chipweinberger