chipweinberger/flutter_blue_plus #718: [Help]: ble keep connect when turning the app off and on

nagosoo, 2023-12-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.30.3

### Flutter Version

3.16.2

### What OS?

Android

### OS Version

android 12

### Bluetooth Module

ESP32-D0WD-V3

### What is your problem?

Hi. I'm using your library well.
I have a question.
I still want to stay connected to the device when I turn the app off and on after connecting the ble device.
I did 'autoConnect:true' from the 'connect' function.
But when I tested it, when I connected it to the ble device and turned it off and off the app, it was disconnected from the device.

Is there any way I can still stay connected to the device when I connect the ble device and turn the app off and on?
I'd love to know how to do this on both Android and IOS.

### Logs

```shell
nothing
```

chipweinberger, 2023-12-06:
not sure tbh

Rajan-Metaliya, 2023-12-06:
Recently I have been working on something similar. For my use case, I have used [flutter_foreground_task](https://pub.dev/packages/flutter_foreground_task) To stay connected and receive or send data between devices.

chipweinberger, 2023-12-14:
iOS documentation here: 

https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1

chipweinberger, 2023-12-14:
added a section to the README.

closed 2023-12-14 by chipweinberger

Mohd-ajmal, 2024-01-20:
> Recently I have been working on something similar. For my use case, I have used [flutter_foreground_task](https://pub.dev/packages/flutter_foreground_task) To stay connected and receive or send data between devices.
> 
> Make sure you ask for all the permissions before starting the foreground service. Because android does not allow to access permissions while app is in foreground.

HI @Rajan-Metaliya, I have some question, can we do this keep our device stay connected for 5 days because it's essential for my app to monitor heart rate continuously and i implemented all essential permission when i close the app, the app keeps running but my ble device connection get loosed.

Rajan-Metaliya, 2024-01-20:
> HI @Rajan-Metaliya, I have some question, can we do this keep our device stay connected for 5 days because it's essential for my app to monitor heart rate continuously and i implemented all [...]

Hi @Mohd-ajmal 
I have tested continuously in foreground for  maximum 15-16 hours and it work as expected. Regarding loosed device connection you can try reconnecting to device on foreground itself. I have done the same and it works for me.