chipweinberger/flutter_blue_plus #719: [Bug]:  write Characteristic always throw a timeout error on Android 9

elliotPopina, 2023-12-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

5.2.0

### Flutter Version

3.16.0

### What OS?

Android

### OS Version

9 (api 28)

### Bluetooth Module

i don't know

### What is your problem?

On Android 9 write Characteristics always throw a timeout error 

I have tried my code with Android 12, and i have no error. 
I have try to set the write timeout to 100 and after 100 secondes the same error is throw 

I will check tomorrow if i have the same error with the example app and update the issue

### Logs

```shell
D/[FBP-Android]( 6818): [FBP] onServicesDiscovered: count: 5 status: 0
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
I/chatty  ( 6818): uid=10156(com.popina.app.dev) Binder:6818_3 identical 32 lines
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/BluetoothGatt( 6818): onConnectionUpdated() - Device=04:7F:0E:98:90:D9 interval=36 latency=0 timeout=500 status=0
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/BluetoothGatt( 6818): onConnectionUpdated() - Device=04:7F:0E:98:90:D9 interval=9 latency=0 timeout=512 status=0
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/[FBP-Android]( 6818): [FBP] onMethodCall: discoverServices
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/BluetoothGatt( 6818): discoverServices() - device: 04:7F:0E:98:90:D9
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/BluetoothGatt( 6818): onSearchComplete() = Device=04:7F:0E:98:90:D9 Status=0
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
I/chatty  ( 6818): uid=10156(com.popina.app.dev) Binder:6818_3 identical 4 lines
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/[FBP-Android]( 6818): [FBP] onServicesDiscovered: count: 5 status: 0
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
I/chatty  ( 6818): uid=10156(com.popina.app.dev) Binder:6818_3 identical 32 lines
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/[FBP-Android]( 6818): [FBP] onMethodCall: getAdapterState
D/[FBP-Android]( 6818): [FBP] onMethodCall: writeCharacteristic
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
I/chatty  ( 6818): uid=10156(com.popina.app.dev) identical 4 lines
2
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/[FBP-Android]( 6818): [FBP] onCharacteristicRead: ff02 status: 2
D/BluetoothDevice( 6818): mAddress: 04:7F:0E:98:90:D9
D/[FBP-Android]( 6818): [FBP] onMethodCall: getAdapterState
```

elliotPopina, 2023-12-07:
Hi have the same log with the example app : 
```dart
I/flutter ( 8500): [FBP] <writeCharacteristic> args: {remote_id: 04:7F:0E:98:90:D9, service_uuid: ff00, secondary_service_uuid: null, characteristic_uuid: ff02, write_type: 1, allow_long_write: 0, value: 2296946e}
D/[FBP-Android]( 8500): [FBP] onMethodCall: writeCharacteristic
D/BluetoothDevice( 8500): mAddress: 04:7F:0E:98:90:D9
I/chatty  ( 8500): uid=10157(com.lib.flutter_blue_plus_example) identical 4 lines
D/BluetoothDevice( 8500): mAddress: 04:7F:0E:98:90:D9
I/flutter ( 8500): [FBP] <writeCharacteristic> result: true
I/flutter ( 8500): [FBP] <getAdapterState> args: null
D/[FBP-Android]( 8500): [FBP] onMethodCall: getAdapterState
D/BluetoothDevice( 8500): mAddress: 04:7F:0E:98:90:D9
D/[FBP-Android]( 8500): [FBP] onCharacteristicRead: ff02 status: 2
D/BluetoothDevice( 8500): mAddress: 04:7F:0E:98:90:D9
I/flutter ( 8500): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter ( 8500): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: ff00, success: 1, remote_id: 04:7F:0E:98:90:D9, error_code: 0, characteristic_uuid: ff02, value: 2296946e}
```
I think your right, the device has a bug, i will try with another device with Android 9

elliotPopina, 2023-12-07:
This is the response from another device (same constructor) and same Android version : 
```dart
I/flutter ( 4488): [FBP] <writeCharacteristic> args: {remote_id: 04:7F:0E:98:90:D9, service_uuid: ff00, secondary_service_uuid: null, characteristic_uuid: ff02, write_type: 1, allow_long_write: 0, value: e3db504d}
D/[FBP-Android]( 4488): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android]( 4488): [FBP] onCharacteristicWrite: ff02 status: 0
I/flutter ( 4488): [FBP] <writeCharacteristic> result: true
I/flutter ( 4488): [FBP] <getAdapterState> args: null
D/[FBP-Android]( 4488): [FBP] onMethodCall: getAdapterState
I/flutter ( 4488): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: ff00, success: 1, remote_id: 04:7F:0E:98:90:D9, error_code: 0, characteristic_uuid: ff02, value: e3db504d}
I/flutter ( 4488): [FBP] <getAdapterState> result: {adapter_state: 4}
```
Everything is fine with this one, the first device does have a problem.
Thanks for your help :)

chipweinberger, 2023-12-07:
> D/[FBP-Android]( 8500): [FBP] onCharacteristicRead: ff02 status: 2

`status: 2` == `BluetoothGatt.GATT_READ_NOT_PERMITTED`

maybe relevant

chipweinberger, 2023-12-14:
closing due to inactivity. This seems like a problem with your device.

closed 2023-12-14 by chipweinberger