chipweinberger/flutter_blue_plus #720: [Help]: Implement Reconnect Device

rivella50, 2023-12-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.30.3

### Flutter Version

3.13.6

### What OS?

All

### OS Version

Doesn't matter

### Bluetooth Module

Bosch PLR 30 C

### What is your problem?

In our mobile app we use a bluetooth sdk we had to create a separate package for in order to reveive values from the mentioned bluetooth device.
Now one of the goals is that the app should automatically reconnect the last connected device if bluetooth is enabled again on that device.
By using the mentioned sdk we are notified about a bluetooth disable action but not the opposite.
Now we had in mind to start scanning for devices as soon the app detects that there is no device connected as long as that device turns on bluetooth or the user connects another device. Scanning could be done by our imported bluetooth package or by using your plugin (where we could add filters).
Now my questions are:

1. Does it make sense having scanning for devices active all the time (performance of the app, energy consumption) in case of no connected device, or do you see any other option (perhaps only activate scanning in short intervals by using a Timer)?
2. How does your plugin's feature `autoConnect:true` work internally (perhaps we could adapt its idea)?
3. Could there be disruptions when we would use 2 different bluetooth packages at the same time (e.g. scanning for devices could potentially overlap)?

Thank you very much for your answers.

### Logs

```shell
Not needed
```

chipweinberger, 2023-12-07:
just use `autoConnect:true`, it is efficient. Android & iOS  use background scans.

If you want to know how it works, just search the codebase for "autoConnect"

chipweinberger, 2023-12-07:
> Could there be disruptions when we would use 2 different bluetooth packages at the same time

Ya I don't think it would work. You can try.

chipweinberger, 2023-12-07:
> Does it make sense having scanning for devices active all the time 

bad idea. too much power.

rivella50, 2023-12-07:
Ok, great, thanks for answering.
The problem is that i cannot use your plugin for the connection part, since i won't get the measurement values from the bluetooth device.
But i will check then how your plugin implements `autoConnect`. Hopefully we can adapt the idea for the integrated bluetooth package.

closed 2023-12-07 by chipweinberger