chipweinberger/flutter_blue_plus #722: [Bug]: Not able to read/write to bluetooth device in VIVO V21 phone

Narayanhegde14, 2023-12-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.12.13

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

Funtouch OS 13 Global

### Bluetooth Module

nordic nrf52840

### What is your problem?

When we do .connect(); In the state listener state is connected after that when I tried writing and reading I'm not receiving any response.

### Logs

```shell
No errors.
```

chipweinberger, 2023-12-09:
update to newer fbp version

Narayanhegde14, 2023-12-09:
> update to newer fbp version
> 
> try example app

```
{disconnect_reason_code: 133, disconnect_reason_string: ANDROID_SPECIFIC_ERROR, remote_id: *****************, connection_state: 0}
```

chipweinberger, 2023-12-09:
ANDROID_SPECIFIC_ERROR is tricky

see readme

chipweinberger, 2023-12-11:
1.12.13 is not supported anymore

also, I would need logs to help you

chipweinberger, 2023-12-14:
closing due to inactivity. Please reopen a new issue with logs if you still have problems.

closed 2023-12-14 by chipweinberger