chipweinberger/flutter_blue_plus #723: [Bug]: Not able to write to the bluetooth device in Xiaomi phone

huubao2309, 2023-12-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.30.3

### Flutter Version

3.13.9

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

LoRa Wireless Stick Lite V3 /433/470-510/863/868-915/928 Hz ESP32 Lora Development Board SX1276 ESP32-PICO-D4 Esp32

### What is your problem?

Not able to write to the Bluetooth device in Xiaomi phone (Redmi Note 11)

### Logs

```shell
FlutterBluePlusException | writeCharacteristic | android-code: 19 | UNKNOWN_GATT_ERROR (19)
```

chipweinberger, 2023-12-10:
what is the characteristic uuid causing the error?

chipweinberger, 2023-12-14:
closing due to inactivity. cannot help you without more details.

closed 2023-12-14 by chipweinberger