chipweinberger/flutter_blue_plus #724: [Help]: Getting state errors on Notify characteristics after reloading screen

kbessemer, 2023-12-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.17.6

### Flutter Version

3.13.7

### What OS?

Android

### OS Version

12

### Bluetooth Module

TI

### What is your problem?

What happens is when I initially run my app, I scan for BLE devices, then connect to one and my BLE device is simulating having slave devices connected to it. So I then choose a slave device to read/write data to/from. When I initially load my dashboard screen, the slave dashboard, everything is loading and working properly... Notifications come in like this with no error:

I/flutter ( 8570): [FBP] [[ OnCharacteristicReceived ]] result: {secondary_service_uuid: null, error_string: GATT_SUCCESS, service_uuid: feed1201-0451-4000-f00d-01140e3e79ae, success: 1, remote_id: 80:6F:B0:1E:FE:BB, error_code: 0, characteristic_uuid: feed0011-0451-4000-f00d-01140e3e79ae, value: a6}
I/flutter ( 8570): RSSI Received value: [166]
I/flutter ( 8570): -90
I/flutter ( 8570): Setting RSSI Value
I/flutter ( 8570): [FBP] [[ OnCharacteristicReceived ]] result: {secondary_service_uuid: null, error_string: GATT_SUCCESS, service_uuid: feed1201-0451-4000-f00d-01140e3e79ae, success: 1, remote_id: 80:6F:B0:1E:FE:BB, error_code: 0, characteristic_uuid: feed0030-0451-4000-f00d-01140e3e79ae, value: 80}
I/flutter ( 8570): Power Status Received value: [128]

But if I navigate away from my "slave dashboard" and back to the slave list, then choose to reconnect to a slave (any of them), I get errors on every notification characteristic coming in... it's a setstate error and the app seems to be functioning properly but the error still appears.

Here is the error coming in on a notify:

E/flutter ( 8570): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: setState() called after dispose(): _DashboardViewState#98f53(lifecycle state: defunct, not mounted)
E/flutter ( 8570): This error happens if you call setState() on a State object for a widget that no longer appears in the widget tree (e.g., whose parent widget no longer includes the widget in its build). This error can occur when code calls setState() from a timer or an animation callback.
E/flutter ( 8570): The preferred solution is to cancel the timer or stop listening to the animation in the dispose() callback. Another solution is to check the "mounted" property of this object before calling setState() to ensure the object is still in the tree.
E/flutter ( 8570): This error might indicate a memory leak if setState() is being called because another object is retaining a reference to this State object after it has been removed from the tree. To avoid memory leaks, consider breaking the reference to this object during dispose().
E/flutter ( 8570): #0      State.setState.<anonymous closure> (package:flutter/src/widgets/framework.dart:1102:9)
E/flutter ( 8570): #1      State.setState (package:flutter/src/widgets/framework.dart:1137:6)
E/flutter ( 8570): #2      _DashboardViewState.GetPowerStatus.<anonymous closure>.<anonymous closure> (package:drc/view/dashboard.dart:405:15)
E/flutter ( 8570): #3      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #4      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #5      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter ( 8570): #6      _SyncBroadcastStreamController._sendData (dart:async/broadcast_stream_controller.dart:377:25)
E/flutter ( 8570): #7      _BroadcastStreamController.add (dart:async/broadcast_stream_controller.dart:244:5)
E/flutter ( 8570): #8      _AsBroadcastStreamController.add (dart:async/broadcast_stream_controller.dart:474:11)
E/flutter ( 8570): #9      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #10     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #11     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter ( 8570): #12     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:784:19)
E/flutter ( 8570): #13     _StreamController._add (dart:async/stream_controller.dart:658:7)
E/flutter ( 8570): #14     _StreamController.add (dart:async/stream_controller.dart:606:5)
E/flutter ( 8570): #15     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #16     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #17     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter ( 8570): #18     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter ( 8570): #19     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
E/flutter ( 8570): #20     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #21     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #22     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
[... 112 lines cut ...]
E/flutter ( 8570): #40     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #41     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #42     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter ( 8570): #43     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter ( 8570): #44     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
E/flutter ( 8570): #45     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #46     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #47     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter ( 8570): #48     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter ( 8570): #49     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
E/flutter ( 8570): #50     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #51     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #52     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter ( 8570): #53     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter ( 8570): #54     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
E/flutter ( 8570): #55     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter ( 8570): #56     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter ( 8570): #57     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
E/flutter ( 8570): #58     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
E/flutter ( 8570): #59     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
E/flutter ( 8570): #60     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter ( 8570): #61     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
E/flutter ( 8570): 
I/flutter ( 8570): Power Status Received value: [0]
```

chipweinberger, 2023-12-10:
thanks for reporting 

we just need to check for `mounted` before we call `setState`

kbessemer, 2023-12-10:
Hey Chip,

Forgive me but I have never done a PR, not sure how to proceed

kbessemer, 2023-12-10:
I added if(mounted) before the setState calls and now I am receiving this error

I also changed my navigation from Navigator.pop to Navigator.pushReplacement, which seemed to have no effect

```
E/flutter (13279): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: 'package:flutter/src/widgets/framework.dart': Failed assertion: line 4833 pos 12: '_lifecycleState != _ElementLifecycle.defunct': is not true.
E/flutter (13279): #0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:51:61)
E/flutter (13279): #1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:40:5)
E/flutter (13279): #2      Element.markNeedsBuild (package:flutter/src/widgets/framework.dart:4833:12)
E/flutter (13279): #3      State.setState (package:flutter/src/widgets/framework.dart:1158:15)
E/flutter (13279): #4      _DashboardViewState.GetRSSI.<anonymous closure>.<anonymous closure> (package:drc/view/dashboard.dart:459:15)
E/flutter (13279): #5      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #6      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #7      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #8      _SyncBroadcastStreamController._sendData (dart:async/broadcast_stream_controller.dart:377:25)
E/flutter (13279): #9      _BroadcastStreamController.add (dart:async/broadcast_stream_controller.dart:244:5)
E/flutter (13279): #10     _AsBroadcastStreamController.add (dart:async/broadcast_stream_controller.dart:474:11)
E/flutter (13279): #11     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #14     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:784:19)
E/flutter (13279): #15     _StreamController._add (dart:async/stream_controller.dart:658:7)
E/flutter (13279): #16     _StreamController.add (dart:async/stream_controller.dart:606:5)
E/flutter (13279): #17     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #18     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #19     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #20     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (13279): #21     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
E/flutter (13279): #22     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (13279): #23     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #24     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #25     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #26     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (13279): #27     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
E/flutter (13279): #28     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (13279): #29     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #30     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #31     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #32     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (13279): #33     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
E/flutter (13279): #34     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (13279): #35     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #36     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #37     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #38     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (13279): #39     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
E/flutter (13279): #40     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (13279): #41     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #42     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #43     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #44     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
E/flutter (13279): #45     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
E/flutter (13279): #46     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
E/flutter (13279): #47     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
E/flutter (13279): #48     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
E/flutter (13279): #49     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
E/flutter (13279): #50     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
[... cut ...]
E/flutter (13279): #67     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
E/flutter (13279): #68     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
E/flutter (13279): #69     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
E/flutter (13279): #70     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter (13279): #71     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
E/flutter (13279): 
I/flutter (13279): Power Status Received value: [128]
```

kbessemer, 2023-12-10:
Alright I found the issue... I was not cancelling the StreamSubscription when navigating away from the screen.

See here I am assigning the lastValueStream to sensorStatusSubscription

```
StreamSubscription? sensorStatusSubscription;
```

```
Future<void> GetSensorStatus() async {
    if (cancel) {
      return;
    }
    try {
      sensorStatusChar.setNotifyValue(true).then((_) {
        if (sensorStatusChar.isNotifying) {
          sensorStatusSubscription = sensorStatusChar.lastValueStream.listen((result) {
            if (cancel) {
              return;
            }
            // Handle the received value here
            int value = result[0];
            String binaryString = value.toRadixString(2);
            print('Sensor Status Received value: ${result}');
            print("Current Status Count: " + currentStatusCount.toString());
            if (mounted) {
              setState(() {
                if (currentStatusCount == 0) {
                  print("Setting Connection Status");
                  try {
                    connectionStatus = binaryString[0];
                  } catch (e) {
                    print("Error setting connectionStatus: " + e.toString());
                  }
                } else if (currentStatusCount == 1) {
                  print("Setting Battery Status");
                  try {
                    batteryStatus = binaryString[0];
                  } catch (e) {
                    print("Error setting batteryStatus: " + e.toString());
                  }
                } else if (currentStatusCount == 2) {
                  print("Setting Solar Status");
                  try {
                    solarStatus = binaryString[0];
                  } catch (e) {
                    print("Error setting solarStatus: " + e.toString());
                  }
                } else if (currentStatusCount == 3) {
                  print("Setting Feed Status");
                  try {
                    feedStatus = binaryString[0];
                  } catch (e) {
                    print("Error setting feedStatus: " + e.toString());
                  }
                } else if (currentStatusCount == 6) {
                  print("Setting Error Status");
                  try {
                    errorStatus = binaryString[0];
                  } catch (e) {
                    print("Error setting errorStatus: " + e.toString());
                  }
                }
                currentStatusCount += 1;
                if (currentStatusCount > 7) {
                  currentStatusCount = 0;
                }
              });
            }
          });
        }
      });
    } catch(e) {
      print("Error in Sensor Status");
      print(e);
    }
  }
```

```
When navigating away from the screen I am simply using:
sensorStatusSubscription.cancel()
```

chipweinberger, 2023-12-11:
yes, that's just a normal flutter error.

closed 2023-12-11 by chipweinberger