chipweinberger/flutter_blue_plus #725: [Help]: I can't get the datas at first time.

100burn, 2023-12-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.1.13

### Flutter Version

3.3.10

### What OS?

iOS

### OS Version

iOS 17/iPhone 15

### Bluetooth Module

Bluetooth LE 4.0

### What is your problem?

I've been using FlutterBluePlus well so far.
But recently the code isn't working very well.

The code is simple.
Press the ‘Scan’ button to scan.
Even if you press the button once, the data is not read well at first.
So, I press the button and then immediately press it again to end the scan. 
Then press Scan again for the function to work.

 Instead of pressing the button 3 times,
Is there a way to scan with just one click?

My code is below.

-----------------------------------------------
bool isScanning = false;
late StreamSubscription<List<ScanResult>> _scanResultsSubscription;

void scan() async {

if (isScanning == false) {
 isScanning = true;

await FlutterBluePlus.startScan(scanMode: ScanMode.lowLatency,
    timeout: const Duration(seconds: 5));
_scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
  // do something with scan results

  setState(() {
    scanResultList = results;
  });

  for (ScanResult s in results) {

int i1 = s.advertisementData
    .manufacturerData[s.advertisementData.manufacturerData.keys.first]!
    .elementAt(0);
print(I);

//   String name = s.device.advName;
String name = s.device.localName;

print(name);

}
});
}
}

<Widget>
ElevatedButton(
  onPressed: scan,
  style: ButtonStyle(
    backgroundColor: MaterialStateProperty.all(
        Color(0xFFFDCC2D)),
  ),
  child: Text(
    isScanning ? ‘scanning..' : ‘Start to scan',
    style: TextStyle(color: Colors.black),
  ),
)

### Logs

```shell
I have to push the button 3 times to get the datas from bluetooth devices.
```

chipweinberger, 2023-12-11:
update your FBP version

100burn, 2023-12-11:
> update your FBP version

I will upgrade to 1.30.3.
Thank you.

chipweinberger, 2023-12-14:
reopen if you still have issues on new version.

closed 2023-12-14 by chipweinberger