chipweinberger/flutter_blue_plus #726: [Help]: Sending 'chunks' over 5 bytes leads to data loss

trabulium, 2023-12-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.29.6

### Flutter Version

3.16.0

### What OS?

All

### OS Version

Android 10 / OS 17

### Bluetooth Module

Nina B406 (nRF52833)

### What is your problem?

If this isn't appropriate, just close the ticket, I understand.

I'm relatively new to both Flutter and Embedded but many years as a web dev (fwiw).

I don't necessarily believe this is a flutter_blue_plus issue but I'm just looking for general direction here in helping to debug. I don't have GDB access on the board unfortunately. Everything works fine when I send small amounts of data. As soon as I start sending approx. > 500 bytes (maximum 650 bytes), I start to lose data. My chunk size matters a lot. The larger the chunk size, the more data gets lost when I verify bytes received. ie: making a chunk size of 20 bytes, bytes sent vs bytes received will be ~20% loss. Chunk size of 8, I will have a 10% loss. If I add a small delay in the sending loop, I have less loss. My BT Nordic SDK is setup with a buffer size of 1024 and MTU of 247.

I am sending a json string like:

```
data = utf8.encode(json.encode(dataToSend));
int offset = 0;
int chunkSize = 5;
bool noResponseRequired = false;

while (offset < data.length) {
      var chunk = data.skip(offset).take(chunkSize).toList();
      offset += chunkSize;
      try {
        await _tx?.write(chunk, withoutResponse: noResponseRequired);
      } catch (e) {
        showToast('Error sending data, trying to reconnect', 'alert');
        _isConnected = false;
        bluetoothManager.connectToDevice(bluetoothManager.currentDevice);
      }
    }
 ```
 
 I set and have verified my MTU is 247, but if I try and use chunkSizes over ~5 or 6, I begin to get data loss / incomplete json arriving on the Board. BLE is supposed to have CRC error checking built in but I'm to the point where I'm beginning to write my own 'chunk verification' but I feel it's over complicated and should be unnecessary.
 
 I'm just looking for generic advice what you would do in this situation or tips on how to better help me debug? 

    
 

### Logs

```shell
No logs worth showing.
```

chipweinberger, 2023-12-11:
seems like a problem with your device. in FBP, we only return from the 'write' command when the device tells us it is complete.

If your MTU is 247, you should get an error if you try to write more than 247.

trabulium, 2023-12-13:
Thanks for the reply, Chip. I agree that it seems like the hardware / device. I'll keep debugging away on the Hardware side of things.

closed 2023-12-14 by chipweinberger

trabulium, 2023-12-20:
I thought I would follow up here for anybody experiencing similar in the future. Our issue was on the device. Atmel / Microchip stopped using the GCC optimisation flag "-Og" in their IDE when they switched to MPLabX IDE. Using this optimisation flag has allowed us to increase our throughput >10x with no data loss.