chipweinberger/flutter_blue_plus #727: [Bug]: Data doesn't seem to flow in every once in a while from notification type characteristics

MrCsabaToth, 2023-12-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.30.3

### Flutter Version

3.18.0-0.1.pre

### What OS?

Android

### OS Version

12

### Bluetooth Module

various

### What is your problem?

Scan and connection works fine. My app signs up to receive data from devices via the `lastValueStream` way. Lately I started to experience randomly (so I cannot reproduce this always), but after the `setNotifyValue(true)` on the particular characteristic which supposed to emit measurements the data doesn't start to flow, the device acts mute. However I see that the devices have power, and action is going on and data supposed to flow in. I started to experience this only lately.

### Logs

```shell
I only have my application specific proprietary logs, I connect to these fitness machines in commercial spaces and it's quite hard to do a LogCat (as soon as owners see my laptop and me debugging they start to shit brix and fear that I screw up their stuff, happened with me multiple times). On app level I haven't pinpointed anything critical yet. I'll keep monitoring the situation and come back with a log if I'll able to obtain any. My main goal with this issue is to survey if anyone else noticed such thing recently, or is it just me. I haven't changed critical core functions around this for a while in my app. The most recent change is to upgrade to `3.18.0-0.1.pre` beta channel Flutter. Not sure if that can affect anything. I'm using the same phone and same fitness machines.
```

chipweinberger, 2023-12-11:
I can't think of anything that would cause these symptoms in FBP.

FBP is just a wrapper around iOS and Android. When the data comes in, we forward it to the stream.

chipweinberger, 2023-12-14:
closing this issue because it is unlikely to be something FBP can fix.

closed 2023-12-14 by chipweinberger