chipweinberger/flutter_blue_plus #729: [Help]: Can't find Rasperry Pi Zero 2 Device

MertSaygili, 2023-12-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

0.4.0

### Flutter Version

3.16.0

### What OS?

Android

### OS Version

13

### Bluetooth Module

Rasperry Pi Zero 2 

### What is your problem?

After startScan() function i cannot find rasperry pi zero 2 device. Bu i can find other devices. I can share my code if needed.

### Logs

```shell
No logs
```

chipweinberger, 2023-12-13:
it's probably bluetooth classic and not ble

MertSaygili, 2023-12-14:
Raspberry pi zero 2 supports ble.

chipweinberger, 2023-12-14:
can you find it using other scanner apps?

MertSaygili, 2023-12-14:
I tried using another BLE scanner app ([app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)). Initially, I couldn't detect the Raspberry Pi. Subsequently, I installed BlueZ on the Raspberry Pi and followed the commands outlined in [this article](https://punchthrough.com/creating-a-ble-peripheral-with-bluez/). Now, I can successfully identify the Raspberry Pi device using the scanDevice() function. However, there is something that I did not understand in the article:

```
When prompted if this should be the primary service, say yes.

Then create our characteristics:

[bluetooth]# register-characteristic 0x1234 read

Enter the initial value you want for that characteristic:

[bluetooth]# register-characteristic 0x4567 read,write
```

What should I write as the initial value for the 'read' characteristic? Additionally, how can I enable the read, write, and notify features on the Raspberry Pi?"

Thanks for your fast response :)

chipweinberger, 2023-12-14:
I'm not familiar with BlueZ, sorry I cannot help!

I think you can set it to anything you want.

closed 2023-12-14 by chipweinberger