chipweinberger/flutter_blue_plus #730: [Bug]: Upgrade from 1.20.4 to 1.30.4 breaks ios build

nichobbs, 2023-12-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.30.4

### Flutter Version

3.16.4

### What OS?

iOS

### OS Version

Sonoma

### Bluetooth Module

Unknown, iPhone 13

### What is your problem?

flutter build ios gives an error after upgrading to version 1.30.4 from an earlier 1.20.4 version. 1.29.13 appears ok.

Semantic Issue (Xcode): Use of undeclared identifier 'autoConnect'
/Users/nic/.pub-cache/hosted/pub.dev/flutter_blue_plus-1.30.4/ios/Classes/FlutterBluePlusPlugin.m:350:35

### Logs

```shell
flutter build ios                                                                                          
Resolving dependencies... 
  _fe_analyzer_shared 61.0.0 (65.0.0 available)
  analyzer 5.13.0 (6.3.0 available)
  built_value 8.7.0 (8.8.1 available)
  dart_style 2.3.2 (2.3.4 available)
  drift 2.12.1 (2.14.1 available)
  intl 0.18.1 (0.19.0 available)
  matcher 0.12.16 (0.12.16+1 available)
  material_color_utilities 0.5.0 (0.8.0 available)
  meta 1.10.0 (1.11.0 available)
  package_info_plus 4.2.0 (5.0.1 available)
  path 1.8.3 (1.9.0 available)
  petitparser 5.4.0 (6.0.2 available)
  test 1.24.9 (1.25.0 available)
  test_api 0.6.1 (0.7.0 available)
  test_core 0.5.9 (0.6.0 available)
  uuid 4.2.2 (4.3.1 available)
  vm_service 13.0.0 (14.0.0 available)
  web 0.3.0 (0.4.0 available)
  xml 6.3.0 (6.5.0 available)
Got dependencies!
19 packages have newer versions incompatible with dependency constraints.
Try `flutter pub outdated` for more information.
Building <appid> for device (ios-release)...
Automatically signing iOS for device deployment using specified development team in Xcode project: <signing key>
Running pod install...                                           2,232ms
Running Xcode build...                                                  
Xcode build done.                                            6.5s
Failed to build iOS app
Semantic Issue (Xcode): Use of undeclared identifier 'autoConnect'
.pub-cache/hosted/pub.dev/flutter_blue_plus-1.30.4/ios/Classes/FlutterBluePlusPlugin.m:350:35
```

kevinconcourse, 2023-12-14:
Facing same issue

agozie, 2023-12-14:
Faced same issue, downgrading to 1.30.3 works.

berkayismus, 2023-12-14:
temporary fix

instead of using in pubspec.yaml;
flutter_blue_plus: ^1.30.4

use;
flutter_blue_plus: 1.30.3

this works for me

chipweinberger, 2023-12-14:
fixed 1.30.5

closed 2023-12-14 by chipweinberger