chipweinberger/flutter_blue_plus #733: [Bug]:  Failed assertion: line 90 pos 12: '(mtu == null) || !autoConnect': mtu and auto connect are incompatible

doguhan2222, 2023-12-18:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

flutter_blue_plus: ^1.30.6

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

13

### Bluetooth Module

Esp32

### What is your problem?

I have always used the same code and the same ESP32 device in 4 different projects for 4 different devices. They all have the same version of blue_plus. But in the last application I made, I cannot connect to the device when `autoconnect = true`. It gives this error. If I set `autoconnect = false`, I can connect to the device successfully.

This is my code for connecting to the device:

```dart
await BaglanilanBluetoothCihaz.BagliBluetoothDevice!.connect(
  autoConnect: true,
  timeout: const Duration(seconds: 15),
);

### Logs

```shell
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: 'package:flutter_blue_plus/src/bluetooth_device.dart': Failed assertion: line 90 pos 12: '(mtu == null) || !autoConnect': mtu and auto connect are incompatible
E/flutter (14415): #0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:51:61)
E/flutter (14415): #1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:40:5)
E/flutter (14415): #2      BluetoothDevice.connect (package:flutter_blue_plus/src/bluetooth_device.dart:90:12)
E/flutter (14415): #3      CihazlarEkraniRepository.baglantiKur (package:xxx/Repository/CihazlarEkraniRepository.dart:76:17)
E/flutter (14415): #4      CihazlarEkraniViewModel.cihazaBaglan (package:boykilo/Viewmodel/CihazlarEkraniViewModel.dart:33:67)
E/flutter (14415): #5      _CihazlarEkraniState.build.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:boykilo/Sayfalar/CihazlarEkrani.dart:292:67)
E/flutter (14415): <asynchronous suspension>
E/flutter (14415):
```

chipweinberger, 2023-12-18:
set mtu to null

closed 2023-12-18 by chipweinberger

samnbd, 2023-12-27:
This is happening to me on iOS even when setting mtu to null

doguhan2222, 2023-12-27:
> This is happening to me on iOS even when setting mtu to null

The same problem still persists. He said and left without any explanation. He was really zero helpful in terms of help.

doguhan2222, 2023-12-27:
> This is happening to me on iOS even when setting mtu to null

Does the same problem persist in release mode? Have you tried it ?. Sometimes the problem disappears on its own in release mode.

samnbd, 2023-12-27:
@doguhan2222 thank you very much, the issue as you suggested is gone when running the app in release mode on iOS and it also resolved this [issue too ](https://github.com/boskokg/flutter_blue_plus/issues/741)

chipweinberger, 2023-12-28:
fixed 1.31.4