chipweinberger/flutter_blue_plus #734: [Help]: how to addservices for bluetoothgatserver

vishwaanv, 2023-12-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.30.7

### Flutter Version

3.1.5

### What OS?

Android

### OS Version

13

### Bluetooth Module

HC-05

### What is your problem?

i want to connect to one device but for that we have to add services to gatt server how to do  that i dint find anything in the documentation ,if i dont add any services it will disconnect automatically

### Logs

```shell
i dont have any logs
```

chipweinberger, 2023-12-19:
## Introduction

FlutterBluePlus is a Bluetooth Low Energy plugin for [Flutter](https://flutter.dev). 

It supports BLE Central Role only (most common). 

If you need BLE Peripheral Role, you should check out [FlutterBlePeripheral](https://pub.dev/packages/flutter_ble_peripheral).

chipweinberger, 2023-12-19:
I recommend you read some tutorials about bluetooth. I dont think you understand Central vs Peripheral very well.

chipweinberger, 2023-12-19:
## ❗ Bluetooth Classic is not supported ❗

 i.e. speakers, headphones, mice, keyboards, gamepads, **Arduino HC-05 & HC-06, and more are not supported.** These all use Bluetooth Classic.

closed 2023-12-19 by chipweinberger