chipweinberger/flutter_blue_plus #735: [Bug]: serviceUuids is blank

erfinbadrian, 2023-12-21:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.30.8

### Flutter Version

3.16.4

### What OS?

Android

### OS Version

android 10

### Bluetooth Module

-

### What is your problem?

i got  serviceUuids is blank

### Logs

```shell
ScanResult{device: BluetoothDevice{remoteId: B0:B2:1C:76:06:FA, platformName: GIB12A5W6C, services: null}, advertisementData: AdvertisementData{advName: GIB12A5W6C, txPowerLevel: 9, connectable: true, manufacturerData: {76: [2, 21, 117, 191, 206, 137, 214, 246, 128, 124, 40, 81, 108, 6, 169, 125, 90, 111, 0, 1, 0, 99, 197]}, serviceData: {}, serviceUuids: []}, rssi: -69, timeStamp: 2023-12-21 11:54:12.767062}
```

chipweinberger, 2023-12-21:
strange. this code has not changed in awhile.

try restarting phone?

please open a PR if you can fix it.

erfinbadrian, 2023-12-21:
Hello @chipweinberger , yes i did restart still not good. any idea?

chipweinberger, 2023-12-21:
I would follow the readme instructions for Debugging, and modify the android java code until it works.

https://github.com/boskokg/flutter_blue_plus/blob/e049cb00b458cd0d3039b2065730fa153ca643fd/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L1849

chipweinberger, 2023-12-21:
oh, also try setting `continuousUpdates` in `startScan` function arguments!

perhaps your device does not immediately advertise its UUIDs?

erfinbadrian, 2023-12-21:
additional information : 
Android 12 also affected 

`ScanResult{device: BluetoothDevice{remoteId: B0:B2:1C:76:06:FA, platformName: GIB12A5W6C, services: null}, advertisementData: AdvertisementData{advName: GIB12A5W6C, txPowerLevel: 9, connectable: true, manufacturerData: {76: [2, 21, 117, 191, 206, 137, 214, 246, 128, 124, 40, 81, 108, 6, 169, 125, 90, 111, 0, 1, 0, 99, 197]}, serviceData: {}, serviceUuids: []}, rssi: -75, timeStamp: 2023-12-21 13:49:56.978846}`

chipweinberger, 2023-12-21:
Are you sure you are properly setting the Service UUID in your device firmware?

Advertisements can usually only by 31 bytes

You have:
- 10 bytes of advName
- 23 bytes of manufacturerData
- ~2 bytes overhead

That is already over the limit. So maybe you are using extended advertising?

erfinbadrian, 2023-12-21:
1.5.5 : 
`scanResult{device: BluetoothDevice{remoteId: B0:B2:1C:76:06:FA, platformName: GIB12A5W6C, services: null}, advertisementData: AdvertisementData{localName: GIB12A5W6C, txPowerLevel: 9, connectable: true, manufacturerData: {76: [2, 21, 117, 191, 206, 137, 214, 246, 128, 124, 40, 81, 108, 6, 169, 125, 90, 111, 0, 1, 0, 99, 197]}, serviceData: {}, serviceUuids: []}, rssi: -72, timeStamp: 2023-12-21 14:31:15.503267}
`

1.18.0 : 

`scanResult{device: BluetoothDevice{remoteId: B0:B2:1C:76:06:FA, platformName: GIB12A5W6C, services: null}, advertisementData: AdvertisementData{localName: GIB12A5W6C, txPowerLevel: 9, connectable: true, manufacturerData: {76: [2, 21, 117, 191, 206, 137, 214, 246, 128, 124, 40, 81, 108, 6, 169, 125, 90, 111, 0, 1, 0, 99, 197]}, serviceData: {}, serviceUuids: []}, rssi: -72, timeStamp: 2023-12-21 14:31:15.503267}`

1.24.0 : 

ScanResult{device: BluetoothDevice{remoteId: B0:B2:1C:76:06:FA, platformName: GIB12A5W6C, services: null}, advertisementData: AdvertisementData{localName: GIB12A5W6C, txPowerLevel: 9, connectable: true, manufacturerData: {76: [2, 21, 117, 191, 206, 137, 214, 246, 128, 124, 40, 81, 108, 6, 169, 125, 90, 111, 0, 1, 0, 99, 197]}, serviceData: {}, serviceUuids: []}, rssi: -86, timeStamp: 2023-12-21 14:35:08.116464}

cc : @chipweinberger

erfinbadrian, 2023-12-21:
Any idea sir? about firmware im sure we are settings it correctly. we also attached screenshot from other ble scanner @chipweinberger

chipweinberger, 2023-12-21:
try removing all other data from the advertisement in firmware

this will confirm if it is an extended advertising issue

chipweinberger, 2023-12-21:
> Are you sure you are properly setting the Service UUID in your device firmware?
> 
> 
> 
> Advertisements can usually only by 31 bytes
> 
> 
> 
> You have:
> 
> - 10 bytes of advName
> 
> - 23 bytes of manufacturerData
> 
> - 1 byte txPowerLevel
> 
> - 1 byte ble flags
> 
> 
> 
> That is already over the limit. So maybe you are using extended advertising?

temporarily delete all of this other data

you have more than 31 bytes already

erfinbadrian, 2023-12-21:
unfortunately we cant remove it. we got the device from the vendor. do you need something else from my side beside that?

chipweinberger, 2023-12-21:
no more ideas ¯\_(ツ)_/¯

chipweinberger, 2023-12-29:
and progress?

MrCsabaToth, 2023-12-30:
I wonder if this can be reproduced with any other type of device which has "extended advertisement". What constitutes to extended advertisement? I'm dealing with wide variety of devices, and serviceUuids are important for me to set them apart regarding what type. So maybe I can check a few devices.

chipweinberger, 2024-01-23:
closing. can't fix it without more assistance from you

closed 2024-01-23 by chipweinberger