chipweinberger/flutter_blue_plus #737: [Help]: When I turn off the device, it still scans, but only the first scan after turning it off can be seen, after that it doesn't scan devices that have been turned off

Han-songyang, 2023-12-22:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

latest

### Flutter Version

3.·0.6

### What OS?

Android

### OS Version

12

### Bluetooth Module

[]

### What is your problem?

When I turn off the device, it still scans, but only the first scan after turning it off can be seen, after that it doesn't scan devices that have been turned off

### Logs

```shell
D/[FBP-Android](24709): [FBP] onMethodCall: startScan
D/BluetoothAdapter(24709): isLeEnabled(): ON
I/flutter (24709): ScanResult{device: BluetoothDevice{remoteId: 34:85:18:92:27:8A, platformName: B3, services: null}, advertisementData: AdvertisementData{advName: B3, txPowerLevel: null, connectable: true, manufacturerData: {36225: [68, 67, 83, 70, 50, 51, 50, 55, 54, 48, 48, 53, 50, 52, 0, 0, 0, 0, 0, 0, 0, 0]}, serviceData: {}, serviceUuids: [00ee, 00ff]}, rssi: -63, timeStamp: 2023-12-22 16:34:28.665434}
I/flutter (24709): ScanResult{device: BluetoothDevice{remoteId: 80:C3:BA:49:F0:5A, platformName: MOMENTUM 4, services: null}, advertisementData: AdvertisementData{advName: MOMENTUM 4, txPowerLevel: null, connectable: true, manufacturerData: {2979: [0, 195, 128], 6: [3, 1, 128, 90, 240, 73, 186, 195, 128, 4, 4, 36]}, serviceData: {}, serviceUuids: [fdce]}, rssi: -42, timeStamp: 2023-12-22 16:34:36.384470}
D/BluetoothLeScanner(24709): onScannerRegistered() - status=0 scannerId=12 mScannerId=0
I actually shut down the second device.
```

chipweinberger, 2023-12-22:
ya thats how its supposed to work!

you can try `removeIfGone`, and `scanResults` vs `onScanResults` to get whatever behavior you want

closed 2023-12-22 by chipweinberger

Han-songyang, 2023-12-22:
I use onScanResults, I think when the device is turned off, it shouldn't be scanned to

chipweinberger, 2023-12-22:
use `removeIfGone`

Han-songyang, 2023-12-22:
I solved the problem, thank you so much!