chipweinberger/flutter_blue_plus #738: [Help]: Zebra ZW 420 is not showing up in the Bluetooth list

hakanekin, 2023-12-22:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.1

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

17.2.1

### Bluetooth Module

iPhone 11 Pro Max

### What is your problem?

The Zebra ZW 420 is not showing up in the Bluetooth list. I can see all the devices around me, but not the printer.

When I scan from my computer, I can see it in the list, and it also appears on Android phones. However, it doesn't show up on iOS (I can't see it in the iphone bluetooth settings either).

This printer is an older model and apparently does not have an **MFI certification**. How can you assist me?

### Logs

```shell
Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, 3.10.6, on Microsoft Windows [Version 10.0.19045.3803], locale tr-TR)
[√] Windows Version (Installed version of Windows is version 10 or higher)
[√] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
[√] Chrome - develop for the web
[√] Visual Studio - develop for Windows (Visual Studio Community 2019 16.11.2)
[√] Android Studio (version 2021.1)
[√] VS Code (version 1.85.1)
[√] Connected device (3 available)
[√] Network resources
```

chipweinberger, 2023-12-22:
try 1.30.9

hakanekin, 2023-12-22:
i dont use any filters, i am running from example and i didnt change anything.

i tried 1.30.9 version right now. its same. i can not see printer in the list

chipweinberger, 2023-12-22:
are you checking other BLE only scanner apps?

the Settings app is not good because it supports bluetooth classic 

you need to check a BLE only scanner app

hakanekin, 2023-12-22:
i tried with BLE Scanner from app store. i can not find bluetooth device.
but i tried with android phone, i could pair and connect without any issue.

this printer does not have mfi certificate. how can i use this printer ?

chipweinberger, 2023-12-23:
you dont need mfi. But I think your printer uses Bluetooth Classic, not BLE.

FBP does not support Bluetooth Classic.

chipweinberger, 2023-12-23:
[image]

You cann see which models have BLE

closed 2023-12-23 by chipweinberger

hakanekin, 2023-12-23:
Model is "RW 420".
do you have any suggestion for my issue ?
do i need to buy a new printer or can i use them ?

chipweinberger, 2023-12-23:
option 1

1. buy printer with BLE (easiest)
2. write flutter plugin for Bluetooth Classic (~10-40hrs of work)