chipweinberger/flutter_blue_plus #739: [Help]: Hardware suggestions for using this package.

ahetawal-p, 2023-12-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

Latest 

### Flutter Version

Latest

### What OS?

Android, iOS

### OS Version

Latest 

### Bluetooth Module

Na

### What is your problem?

Hello,
I am trying to do some exploring with beacons for a project. But was wondering if anyone has suggestions for which kind of hardware/beacons I can use with this package. I found some of them on Amazon, but just not sure if they will really work with this package.
The only requirement for me is that they should work on an iOS app.
Ex: https://www.amazon.com/gp/aw/d/B085XN9B7N/ref=ox_sc_act_image_1?smid=A3LUSRMOBZQLMS&psc=1

Any suggestions of where can I buy the hardware.

Thanks again,

### Logs

```shell
Na
```

chipweinberger, 2023-12-24:
ibeacons require CoreLocation on ios, which fbp does not support 

see readme

i recommend esp32-s3

closed 2023-12-25 by chipweinberger

ahetawal-p, 2023-12-26:
@chipweinberger any idea why it will not work with eddystone broadcasting.
While esp32-s3 will work, I am looking for a more aesthetically complete hardware at a commercial level, if we can just stick them at places, instead setting up the above micro processor with a power supply everywhere...

chipweinberger, 2023-12-26:
not sure you can try it