chipweinberger/flutter_blue_plus #740: [Bug]: Cannot determine adapter state with or after FBP 1.30.7

MrCsabaToth, 2023-12-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.30.7+

### Flutter Version

3.18.0-0.2.pre

### What OS?

Android

### OS Version

12, 13

### Bluetooth Module

Qualcomm SoC

### What is your problem?

With FBP 1.30.7 and above I cannot obtain the adapter state, it times out. For a good while my code is:

    final blueState = await FlutterBluePlus.adapterState.first.timeout(
      const Duration(milliseconds: 4000),
      onTimeout: () => BluetoothAdapterState.off);

After upgrading to 1.30.7 this code times out, and the code concludes to adapter being off. Couldn't pinpoint what code change caused this between 1.30.6 and 1.30.7. I was testing it with the example app:

    if (FlutterBluePlus.isScanningNow) {
      return FloatingActionButton(
        child: const Icon(Icons.stop),
        onPressed: onStopPressed,
        backgroundColor: Colors.red,
      );
    } else {
      return Row(children: [
        FloatingActionButton(child: const Text("DEBUG"), onPressed: () async {
          final firstBlueState = await FlutterBluePlus.adapterState.first.timeout(
            const Duration(milliseconds: 4000),
            onTimeout: () => BluetoothAdapterState.off);
          final lastBlueState = await FlutterBluePlus.adapterState.last.timeout(
            const Duration(milliseconds: 4000),
            onTimeout: () => BluetoothAdapterState.off);
          debugPrint("Last: $firstBlueState $lastBlueState");
        }),
        FloatingActionButton(child: const Text("SCAN"), onPressed: onScanPressed)
      ]);

### Logs

```shell
If I scan despite of the adapter state it comes back with:

    D/BluetoothAdapter(21765): isLeEnabled(): ON
```

zhxst, 2023-12-25:
Same issue here. 
iPhone 11 with iOS 16.5.1
FlutterBluePlus version 1.31.2.
`FlutterBluePlus.adapterState.first` never return the result, neither when listen to the stream.

chipweinberger, 2023-12-25:
the problem is this commit https://github.com/boskokg/flutter_blue_plus/commit/22d7e4a84323823a41d796268fc86fdbe94b38fd

this was an important fix for a perf issue

maybe you can try to fix it?

MrCsabaToth, 2023-12-25:
Based on f036d8717bed7b1c1ace7f979a49f23216980d58 commit I will try `adapterStateNow`.

MrCsabaToth, 2023-12-25:
Is `_BufferStream` used for anything else than the adapter state (I am not at the make right now, just on my phone)? I'll potentially debug some tomorrow.

MrCsabaToth, 2023-12-25:
Thanks for pointing towards #661. Tomorrow's (today's) debug I will also make sure that the broadcast nature remains ok. That becomes important if the application level code listens on a stream multiple times!

BTW I rearchitected my app a long time ago to minimize the number of scans and any calls towards the native bt protocol stack. For example latching on even the adapter isOn state could mean periodic polls. Even though the app becomes less reactive without the listens, in the long run it aggravates the protocol stack less. I've found that the periodic calls in the "background" (quotes because I don't mean technically in the background, but rather just under the hood) over a long application run could cause problems on some devices. Such as not returning scans, needing app restarts.

chipweinberger, 2023-12-25:
I'm not reproducing this issue (tested on macOS)

```
flutter: [FBP] <setLogLevel> args: 5
flutter: adapterState.listen
2023-12-25 10:20:23.758 macos[37695:5334714] [FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
flutter: [FBP] <setLogLevel> result: true
flutter: [FBP] <getAdapterState> args: null
2023-12-25 10:20:23.928 macos[37695:5334714] [FBP-iOS] handleMethodCall: getAdapterState
flutter: [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 4}
flutter: [FBP] <getAdapterState> result: {adapter_state: 4}
flutter: bind b
flutter: create b
flutter: return
flutter: listen 0
flutter: BluetoothAdapterState.on
flutter: scanResults.listen
flutter: bind b
flutter: create b
flutter: return
flutter: listen 0
flutter: scanResults: []
flutter: isScanning.listen
flutter: bind b
flutter: create b
flutter: return
flutter: listen 0
flutter: isScanning: false
```

```
// Helper for 'newStreamWithInitialValue' method for streams.
class _NewStreamWithInitialValueTransformer<T> extends StreamTransformerBase<T, T> {
  /// the initial value to push to the new stream
  final T initialValue;

  /// controller for the new stream
  late StreamController<T> controller;

  /// subscription to the original stream
  late StreamSubscription<T> subscription;

  /// new stream listener count
  var listenerCount = 0;

  _NewStreamWithInitialValueTransformer(this.initialValue);

  @override
  Stream<T> bind(Stream<T> stream) {
    if (stream.isBroadcast) {
      print("bind b");
      return _bind(stream, broadcast: true);
    } else {
      print("bind s");
      return _bind(stream);
    }
  }

  Stream<T> _bind(Stream<T> stream, {bool broadcast = false}) {
    /////////////////////////////////////////
    /// Original Stream Subscription Callbacks
    ///

    /// When the original stream emits data, forward it to our new stream
    void onData(T data) {
      print("data");
      controller.add(data);
    }

    /// When the original stream is done, close our new stream
    void onDone() {
      print("done");
      controller.close();
    }

    /// When the original stream has an error, forward it to our new stream
    void onError(Object error) {
      print("error");
      controller.addError(error);
    }

    /// When a client listens to our new stream, emit the
    /// initial value and subscribe to original stream if needed
    void onListen() {
      print("listen $listenerCount");
      // Emit the initial value to our new stream
      controller.add(initialValue);

      // listen to the original stream, if needed
      if (listenerCount == 0) {
        subscription = stream.listen(
          onData,
          onError: onError,
          onDone: onDone,
        );
      }

      // count listeners of the new stream
      listenerCount++;
    }

    //////////////////////////////////////
    ///  New Stream Controller Callbacks
    ///

    /// (Single Subscription Only) When a client pauses
    /// the new stream, pause the original stream
    void onPause() {
      print("pause");
      subscription.pause();
    }

    /// (Single Subscription Only) When a client resumes
    /// the new stream, resume the original stream
    void onResume() {
      print("resume");
      subscription.resume();
    }

    /// Called when a client cancels their
    /// subscription to the new stream,
    void onCancel() {
      print("cancel $listenerCount");
      // count listeners of the new stream
      listenerCount--;

      // when there are no more listeners of the new stream,
      // cancel the subscription to the original stream,
      // and close the new stream controller
      if (listenerCount == 0) {
        subscription.cancel();
        controller.close();
      }
    }

    //////////////////////////////////////
    /// Return New Stream
    ///

    // create a new stream controller
    if (broadcast) {
      print("create b");
      controller = StreamController<T>.broadcast(
        onListen: onListen,
        onCancel: onCancel,
        sync: true,
      );
    } else {
      print("create s");
      controller = StreamController<T>(
        onListen: onListen,
        onPause: onPause,
        onResume: onResume,
        onCancel: onCancel,
        sync: true,
      );
    }

    print("return");
    return controller.stream;
  }
}
```

```
class _ScanScreenState extends State<ScanScreen> {

  @override
  void initState() {
    super.initState();
    print("scanResults.listen");
    _scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      print("scanResults: $results");
      _scanResults = results;
      if (mounted) {
        setState(() {});
      }
    }, onError: (e) {
      Snackbar.show(ABC.b, prettyException("Scan Error:", e), success: false);
    });
    print("isScanning.listen");
    _isScanningSubscription = FlutterBluePlus.isScanning.listen((state) {
      print("isScanning: $state");
      _isScanning = state;
      if (mounted) {
        setState(() {});
      }
    });
  }
```

```
class _FlutterBlueAppState extends State<FlutterBlueApp> {

  @override
  void initState() {
    super.initState();
    print("adapterState.listen");
    _adapterStateStateSubscription = FlutterBluePlus.adapterState.listen((state) {
      print("$state");
      _adapterState = state;
      if (mounted) {
        setState(() {});
      }
    });
  }
```

chipweinberger, 2023-12-25:
hmm. it does reproduce when using your example code. weird.

```
flutter: bind b
flutter: create b
flutter: return
flutter: listen 0
flutter: bind b
flutter: create b
flutter: return
flutter: listen 0
flutter: Last: BluetoothAdapterState.off BluetoothAdapterState.off
```

```
if (FlutterBluePlus.isScanningNow) {
  return FloatingActionButton(
    child: const Icon(Icons.stop),
    onPressed: onStopPressed,
    backgroundColor: Colors.red,
  );
} else {
  return Row(children: [
    FloatingActionButton(child: const Text("DEBUG"), onPressed: () async {
      final firstBlueState = await FlutterBluePlus.adapterState.first.timeout(
        const Duration(milliseconds: 4000),
        onTimeout: () => BluetoothAdapterState.off);
      final lastBlueState = await FlutterBluePlus.adapterState.last.timeout(
        const Duration(milliseconds: 4000),
        onTimeout: () => BluetoothAdapterState.off);
      debugPrint("Last: $firstBlueState $lastBlueState");
    }),
    FloatingActionButton(child: const Text("SCAN"), onPressed: onScanPressed)
  ]);
```

chipweinberger, 2023-12-25:
so this code works

```
    FlutterBluePlus.adapterState.listen((event) {
      print("event $event");
    });
```

but this code doesnt 

```
    FlutterBluePlus.adapterState.first.then((event) {
      print("first event $event");
    });
```

it makes no sense to me

chipweinberger, 2023-12-25:
the bug looks to be in Dart....

they call listen without setting onData until later, so the first value is missed.

this seems like a big bug...
```
  Future<T> get first {
    _Future<T> future = new _Future<T>();
    StreamSubscription<T> subscription =
        this.listen(null, onError: future._completeError, onDone: () {
      try {
        throw IterableElementError.noElement();
      } catch (e, s) {
        _completeWithErrorCallback(future, e, s);
      }
    }, cancelOnError: true);
    subscription.onData((T value) {
      _cancelAndValue(subscription, future, value);
    });
    return future;
  }
```

chipweinberger, 2023-12-25:
this is a workaround for the Dart bug

```
extension FirstEventExtension<T> on Stream<T> {
  Future<T> get firstEvent {
    var completer = Completer<T>();
    StreamSubscription<T>? subscription;

    subscription = this.listen(
      (data) {
        if (!completer.isCompleted) {
          completer.complete(data);
          subscription?.cancel();
        }
      },
      onError: (error) {
        if (!completer.isCompleted) {
          completer.completeError(error);
          subscription?.cancel();
        }
      },
      onDone: () {
        if (!completer.isCompleted) {
          completer.completeError(StateError('No element'));
        }
      },
      cancelOnError: true,
    );

    return completer.future;
  }
}
```

usage:
```
    FlutterBluePlus.adapterState.firstEvent.then((event) {
      print("first event $event");
    });
```

chipweinberger, 2023-12-25:
filed this bug report: https://github.com/flutter/flutter/issues/140609

chipweinberger, 2023-12-25:
we can fix this in FBP by setting "sync:false" in 

```
      controller = StreamController<T>.broadcast(
        onListen: onListen,
        onCancel: onCancel,
        sync: false,
      ); 
```

chipweinberger, 2023-12-25:
fixed 1.31.3

closed 2023-12-25 by chipweinberger

MrCsabaToth, 2023-12-26:
Wow, this is fascinating. Thanks for the debug. I'll test.

MrCsabaToth, 2023-12-26:
Looks like working, although first tests yielded blank scans. I'm performing the scan a different way than the example. I'm using a controller, I obtain the stream through that, and I also use some throttling. After some fiddling the empty scan fixed itself.

MrCsabaToth, 2023-12-30:
Just a question: even though this is in the example app, should this `sync: true` be removed as well? https://github.com/boskokg/flutter_blue_plus/blob/178032de10ec91d6f08f0ce0d072eb30c847a993/example/lib/utils/utils.dart#L78

chipweinberger, 2023-12-31:
yes. thanks. perhaps you can make a PR

MrCsabaToth, 2023-12-31:
https://github.com/boskokg/flutter_blue_plus/pull/747