chipweinberger/flutter_blue_plus #741: [Bug]: Auto connect not working on iOS

samnbd, 2023-12-27:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.3

### Flutter Version

3.7.11

### What OS?

iOS

### OS Version

17.2.1

### Bluetooth Module

Pavlok device

### What is your problem?

When pairing the device with `device.connect(autoConnect: true, mtu: null)`,  it connects the device and when I move the device out of range it disconnects and re-connects as expected when it comes in range on Android. 
But on iOS, it doesn't re-connect when it comes back in range. and I have to re-connect it manually.
Notice int the logs that the error **(mtu == null) || !autoConnect': mtu and auto connect are incompatible** is being thrown even though, I've set the mtu: null in the connect function.

### Logs

```shell
[FBP-iOS] handleMethodCall: connect
[FBP-iOS] didConnectPeripheral
[FBP-iOS] didDisconnectPeripheral:
[FBP-iOS]   error: The connection has timed out unexpectedly.
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: 'package:flutter_blue_plus/src/bluetooth_device.dart': Failed assertion: line 95 pos 12: '(mtu == null) || !autoConnect': mtu and auto connect are incompatible
#0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:51:61)
#1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:40:5)
#2      BluetoothDevice.connect (package:flutter_blue_plus/src/bluetooth_device.dart:95:12)
#3      FlutterBluePlus._methodCallHandler (package:flutter_blue_plus/src/flutter_blue_plus.dart:431:44)
#4      MethodChannel._handleAsMethodCall (package:flutter/src/services/platform_channel.dart:555:55)
#5      MethodChannel.setMethodCallHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:548:34)
#6      _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:393:35)
#7      _invoke2 (dart:ui/hooks.dart:183:13)
#8      _ChannelCallbackRecord.invoke (dart:ui/channel_buffers.dart:40:5)
#9      _Channel.push (dart:ui/channel_buffers.dart:130:31)
#10     ChannelBuffers.push (dart:ui/channel_buffers.dart:326:17)
#11     PlatformDispatcher._dispatchPlatformMessage (dart:ui/platform_dispatcher.dart:664:22)
#12     _dispatchPlatformMessage (dart:ui/hooks.dart:86:31)
```

samnbd, 2023-12-27:
The auto connect issue on iOS is not working because of the `(mtu == null) || !autoConnect': mtu and auto connect are incompatible` and it only happens when I run the app in debug mode on iOS. But when running the app in release mode auto connect works correctly and the `mtu` error is not thrown

chipweinberger, 2023-12-28:
fixed 1.31.4

closed 2023-12-28 by chipweinberger

samnbd, 2023-12-28:
Fix confirmed...Thanks