chipweinberger/flutter_blue_plus #742: [Help]: After modifying the name of a Bluetooth device, the iOS device still scanAfter modifying the name of a Bluetooth device, the iOS device still scans the old Bluetooth name, while the Android device scans the new name. The nrf Connect software also scans the new Bluetooth name.

guduqiucai, 2023-12-28:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.3

### Flutter Version

3.10.2

### What OS?

iOS

### OS Version

16.4.1

### Bluetooth Module

unknown

### What is your problem?

[Bug]: After modifying the name of a Bluetooth device, the iOS device still scanAfter modifying the name of a Bluetooth device, the iOS device still scans the old Bluetooth name, while the Android device scans the new name. The nrf Connect software also scans the new Bluetooth name.

### Logs

```shell
no
```

chipweinberger, 2023-12-28:
use advName instead of platformName

guduqiucai, 2023-12-28:
for (ScanResult r in results) {
          if (Platform.isAndroid) {
            if (r.device.localName != '' &&
                !blueName.contains(r.device.localName)) {
              blueName.add(r.device.localName);
              if (mounted) {
                setState(() {
                  this._blueDevice.add(r);
                });
              }
            }
          } else {
            if (r.device.advName != '' &&
                !blueName.contains(r.advertisementData.localName)) {
              blueName.add(r.advertisementData.advName);
              if (mounted) {
                setState(() {
                  this._blueDevice.add(r);
                });
              }
            }
          }
        }

The getter 'advName' isn't defined for the type 'BluetoothDevice'.

chipweinberger, 2023-12-28:
its defined here

https://github.com/boskokg/flutter_blue_plus/blob/fb436621b0951b4940d97853e3248f9d6a95b535/lib/src/bluetooth_device.dart#L35

guduqiucai, 2023-12-28:
Will this be uniformly modified in subsequent versions, or can only I modify it locally?

guduqiucai, 2023-12-28:
> Will this be uniformly modified in subsequent versions, or can only I modify it locally?

Or how should I use it?

chipweinberger, 2023-12-28:
```
/// Advertised Named
  ///  - the is the name advertised by the device during scanning
  ///  - it is only available after you scan with FlutterBluePlus
  ///  - it is cleared when the app restarts.
  ///  - not all devices advertise a name
```

closed 2023-12-29 by guduqiucai

GhazanfarU6, 2024-02-26:
i have the same issue some device names are being displayed but some are not advertising their name, so is there any other solution to get their names because in my context its necessary to show device name that are being scanned.

chipweinberger, 2024-02-26:
if your device does not share its name, there is nothing we can do.