chipweinberger/flutter_blue_plus #744: [Help]: problem when scanning devices

micromotores, 2023-12-28:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.4

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

HM10 

### What is your problem?

Good afternoon, I am using the library in my project, when I have migrated from version 1.5.2 to the current version, the list of devices does not appear, I have downloaded the example code and I cannot get it to work either. With version 1.5.2 I had no problems, but when I went to upload the release to the play store the button I used to scan the devices did not work, but the strangest thing is that in debug mode it worked perfectly. Something you can help me with.

### Logs

```shell
Launching lib\main.dart on SM A515F in debug mode...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:64933/TYxTtXJrOkU=/ws
D/[FBP-Android]( 3344): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 3344): [FBP] initializing BluetoothAdapter
I/flutter ( 3344): [FBP] <setLogLevel> args: 5
I/BluetoothAdapter( 3344): BluetoothAdapter() : com.example.last_ble_plus
I/BluetoothAdapter( 3344): STATE_ON
D/[FBP-Android]( 3344): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 3344): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 3344): [FBP] onMethodCall: setLogLevel
I/flutter ( 3344): [FBP] <setLogLevel> result: true
I/flutter ( 3344): [FBP] <getAdapterState> args: null
D/[FBP-Android]( 3344): [FBP] onMethodCall: getAdapterState
I/flutter ( 3344): [FBP] <getAdapterState> result: {adapter_state: 4}
I/Gralloc4( 3344): mapper 4.x is not supported
W/Gralloc3( 3344): mapper 3.x is not supported
I/gralloc ( 3344): Arm Module v1.0
W/Gralloc4( 3344): allocator 4.x is not supported
W/Gralloc3( 3344): allocator 3.x is not supported
I/BLASTBufferQueue( 3344): [SurfaceView[com.example.last_ble_plus/com.example.last_ble_plus.MainActivity]@0#1](f:0,a:0) onFrameAvailable the first frame is available
D/SurfaceView@aea1e1d( 3344):  setAlpha: mUseAlpha = false alpha=1.0
D/SurfaceView@aea1e1d( 3344):  updateSurfaceAlpha: setUseAlpha() is not called, ignored.
I/ViewRootImpl@de8f851[MainActivity]( 3344): Setup new sync id=0
I/ViewRootImpl@de8f851[MainActivity]( 3344): Setting syncFrameCallback
I/ViewRootImpl@de8f851[MainActivity]( 3344): registerCallbacksForSync syncBuffer=false
I/ViewRootImpl@de8f851[MainActivity]( 3344): mWNT: t=0x72c6c4da00 mBlastBufferQueue=0x72c6bf2f00 fn= 1 caller= android.view.ViewRootImpl.lambda$applyTransactionOnDraw$11$android-view-ViewRootImpl:12865 android.view.ViewRootImpl$$ExternalSyntheticLambda1.onFrameDraw:4 android.graphics.HardwareRenderer$FrameDrawingCallback.onFrameDraw:924
I/ViewRootImpl@de8f851[MainActivity]( 3344): Received frameDrawingCallback syncResult=0 frameNum=1.
I/ViewRootImpl@de8f851[MainActivity]( 3344): Setting up sync and frameCommitCallback
D/SurfaceView@aea1e1d( 3344): updateSurfacePosition RenderWorker, frameNr = 1, position = [0, 0, 1080, 2256] surfaceSize = 1080x2256
I/SurfaceView@aea1e1d( 3344): uSP: rtp = Rect(0, 0 - 1080, 2256) rtsw = 1080 rtsh = 2256
I/SurfaceView@aea1e1d( 3344): onSSPAndSRT: pl = 0 pt = 0 sx = 1.0 sy = 1.0
I/SurfaceView@aea1e1d( 3344): aOrMT: ViewRootImpl@de8f851[MainActivity] t = android.view.SurfaceControl$Transaction@cafc4c6 fN = 1 android.view.SurfaceView.-$$Nest$mapplyOrMergeTransaction:0 android.view.SurfaceView$SurfaceViewPositionUpdateListener.positionChanged:1537 android.graphics.RenderNode$CompositePositionUpdateListener.positionChanged:373
I/ViewRootImpl@de8f851[MainActivity]( 3344): mWNT: t=0x72c6c4d900 mBlastBufferQueue=0x72c6bf2f00 fn= 1 caller= android.view.SurfaceView.applyOrMergeTransaction:1455 android.view.SurfaceView.-$$Nest$mapplyOrMergeTransaction:0 android.view.SurfaceView$SurfaceViewPositionUpdateListener.positionChanged:1537
I/BLASTBufferQueue( 3344): [ViewRootImpl@de8f851[MainActivity]#0](f:0,a:0) onFrameAvailable the first frame is available
[... 7 lines cut ...]
D/InputMethodManager( 3344): startInputInner - Id : 0
I/InputMethodManager( 3344): startInputInner - mService.startInputOrWindowGainedFocus
D/InputMethodManager( 3344): startInputInner - Id : 0
D/InsetsController( 3344): onStateChanged: InsetsState: {mDisplayFrame=Rect(0, 0 - 1080, 2400), mDisplayCutout=DisplayCutout{insets=Rect(0, 88 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(512, 0 - 568, 88), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=1080 displayHeight=2400 physicalDisplayWidth=1080 physicalDisplayHeight=2400 density={2.625} cutoutSpec={M 0,0 M 0,12.25101276465640 a 10.60613009248640,10.60613009248640 0 1,0 0,21.21226018497290 a 10.60613009248640,10.60613009248640 0 1,0 0,-21.21226018497290 Z @dp} rotation={0} scale={1.0} physicalPixelDisplaySizeRatio={1.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=96, center=Point(96, 96)}, RoundedCorner{position=TopRight, radius=96, center=Point(984, 96)}, RoundedCorner{position=BottomRight, radius=96, center=Point(984, 2304)}, RoundedCorner{position=BottomLeft, radius=96, center=Point(96, 2304)}]}  mRoundedCornerFrame=Rect(0, 0 - 1080, 2400), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(948, 0 - 1080, 88) rotation=0}, mSources= { InsetsSource: {mType=ITYPE_STATUS_BAR, mFrame=[0,0][1080,88], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_NAVIGATION_BAR, mFrame=[0,2256][1080,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_LEFT_GESTURES, mFrame=[0,0][0,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_RIGHT_GESTURES, mFrame=[1080,0][1080,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_MANDATORY_GESTURES, mFrame=[0,0][1080,124], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_MANDATORY_GESTURES, mFrame=[0,2256][1080,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_LEFT_DISPLAY_CUTOUT, mFrame=[0,0][-100000,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_DISPLAY_CUTOUT, mFrame=[0,0][1080,88], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_RIGHT_DISPLAY_CUTOUT, mFrame=[100000,0][1080,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_DISPLAY_CUTOUT, mFrame=[0,100000][1080,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_TAPPABLE_ELEMENT, mFrame=[0,0][1080,88], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_TAPPABLE_ELEMENT, mFrame=[0,2256][1080,2400], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_IME, mFrame=[0,0][0,0], mVisibleFrame=[0,1339][1080,2400], mVisible=false, mInsetsRoundedCornerFrame=false} } host=com.example.last_ble_plus/com.example.last_ble_plus.MainActivity from=android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:6740
I/ViewRootImpl@de8f851[MainActivity]( 3344): ViewPostIme pointer 0
I/ViewRootImpl@de8f851[MainActivity]( 3344): ViewPostIme pointer 1
I/flutter ( 3344): [FBP] <getSystemDevices> args: null
D/[FBP-Android]( 3344): [FBP] onMethodCall: getSystemDevices
I/flutter ( 3344): [FBP] <getSystemDevices> result: {devices: []}
I/flutter ( 3344): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 8, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android]( 3344): [FBP] onMethodCall: startScan
I/BluetoothAdapter( 3344): STATE_ON
I/BluetoothAdapter( 3344): STATE_ON
I/BluetoothAdapter( 3344): STATE_ON
I/BluetoothAdapter( 3344): STATE_ON
D/BluetoothLeScanner( 3344): Start Scan with callback
D/BluetoothLeScanner( 3344): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
I/flutter ( 3344): [FBP] <startScan> result: true
I/flutter ( 3344): [FBP] <stopScan> args: null
D/[FBP-Android]( 3344): [FBP] onMethodCall: stopScan
I/BluetoothAdapter( 3344): STATE_ON
I/BluetoothAdapter( 3344): STATE_ON
D/BluetoothLeScanner( 3344): Stop Scan with callback
I/flutter ( 3344): [FBP] <stopScan> result: true
```

micromotores, 2023-12-28:
I updated the version of flutter to 3.16.5, I have tested it and it is still the same, I copied the log.

√  Built build\app\outputs\flutter-apk\app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:56149/h5sxuWo-jw0=/ws
D/[FBP-Android]( 4294): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 4294): [FBP] initializing BluetoothAdapter
I/flutter ( 4294): [FBP] <setLogLevel> args: 5
I/BluetoothAdapter( 4294): BluetoothAdapter() : com.example.last_ble_plus
I/BluetoothAdapter( 4294): STATE_ON
D/[FBP-Android]( 4294): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 4294): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 4294): [FBP] onMethodCall: setLogLevel
I/flutter ( 4294): [FBP] <setLogLevel> result: true
I/flutter ( 4294): [FBP] <getAdapterState> args: null
D/[FBP-Android]( 4294): [FBP] onMethodCall: getAdapterState
I/flutter ( 4294): [FBP] <getAdapterState> result: {adapter_state: 4}
I/ViewRootImpl@de8f851[MainActivity]( 4294): performTraversals params={(0,0)(fillxfill) sim={adjust=resize forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030001
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fl=81810100
I/ViewRootImpl@de8f851[MainActivity]( 4294):   pfl=16020040
I/ViewRootImpl@de8f851[MainActivity]( 4294):   vsysui=500
I/ViewRootImpl@de8f851[MainActivity]( 4294):   apr=LIGHT_STATUS_BARS
I/ViewRootImpl@de8f851[MainActivity]( 4294):   bhv=DEFAULT
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fitSides= naviIconColor=0
I/ViewRootImpl@de8f851[MainActivity]( 4294):   sfl=100000}
I/ViewRootImpl@de8f851[MainActivity]( 4294): performTraversals mFirst=false windowShouldResize=false viewVisibilityChanged=false mForceNextWindowRelayout=false params={(0,0)(fillxfill) sim={adjust=resize forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030001
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fl=81810100
I/ViewRootImpl@de8f851[MainActivity]( 4294):   pfl=16020040
I/ViewRootImpl@de8f851[MainActivity]( 4294):   vsysui=500
I/ViewRootImpl@de8f851[MainActivity]( 4294):   apr=LIGHT_STATUS_BARS
I/ViewRootImpl@de8f851[MainActivity]( 4294):   bhv=DEFAULT
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fitSides= naviIconColor=0
I/ViewRootImpl@de8f851[MainActivity]( 4294):   sfl=100000}
I/ViewRootImpl@de8f851[MainActivity]( 4294): updateBlastSurfaceIfNeeded mBlastBufferQueue=0x72bf0d4a00 isSameSurfaceControl=true
I/BLASTBufferQueue( 4294): update, w= 1080 h= 2400 mName = ViewRootImpl@de8f851[MainActivity] mNativeObject= 0x72bf0d4a00 sc.mNativeObject= 0x724c045160 format= -3 caller= android.view.ViewRootImpl.updateBlastSurfaceIfNeeded:2898 android.view.ViewRootImpl.relayoutWindow:9847 android.view.ViewRootImpl.performTraversals:3884 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301
I/ViewRootImpl@de8f851[MainActivity]( 4294): Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)0 dur=13 res=0x1 s={true 0x724c213000} ch=false seqId=0
I/ViewRootImpl@de8f851[MainActivity]( 4294): updateBoundsLayer: t = android.view.SurfaceControl$Transaction@63ea308 sc = Surface(name=Bounds for - com.example.last_ble_plus/com.example.last_ble_plus.MainActivity@0)/@0xe880aa1 frame = 1
I/ViewRootImpl@de8f851[MainActivity]( 4294): mWNT: t=0x724c109100 mBlastBufferQueue=0x72bf0d4a00 fn= 1 caller= android.view.ViewRootImpl.prepareSurfaces:2985 android.view.ViewRootImpl.performTraversals:4233 android.view.ViewRootImpl.doTraversal:3116
I/ViewRootImpl@de8f851[MainActivity]( 4294): reportNextDraw android.view.ViewRootImpl.performTraversals:4438 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301 android.view.Choreographer$CallbackRecord.run:1309
I/Gralloc4( 4294): mapper 4.x is not supported
W/Gralloc3( 4294): mapper 3.x is not supported
I/gralloc ( 4294): Arm Module v1.0
W/Gralloc4( 4294): allocator 4.x is not supported
W/Gralloc3( 4294): allocator 3.x is not supported
I/BLASTBufferQueue( 4294): [SurfaceView[com.example.last_ble_plus/com.example.last_ble_plus.MainActivity]@0#1](f:0,a:0) onFrameAvailable the first frame is available
D/SurfaceView@aea1e1d( 4294):  setAlpha: mUseAlpha = false alpha=1.0
D/SurfaceView@aea1e1d( 4294):  updateSurfaceAlpha: setUseAlpha() is not called, ignored.
I/ViewRootImpl@de8f851[MainActivity]( 4294): Setup new sync id=0
I/ViewRootImpl@de8f851[MainActivity]( 4294): Setting syncFrameCallback
I/ViewRootImpl@de8f851[MainActivity]( 4294): registerCallbacksForSync syncBuffer=false
I/ViewRootImpl@de8f851[MainActivity]( 4294): mWNT: t=0x724c109400 mBlastBufferQueue=0x72bf0d4a00 fn= 1 caller= android.view.ViewRootImpl.lambda$applyTransactionOnDraw$11$android-view-ViewRootImpl:12865 android.view.ViewRootImpl$$ExternalSyntheticLambda1.onFrameDraw:4 android.graphics.HardwareRenderer$FrameDrawingCallback.onFrameDraw:924
I/ViewRootImpl@de8f851[MainActivity]( 4294): Received frameDrawingCallback syncResult=0 frameNum=1.
I/ViewRootImpl@de8f851[MainActivity]( 4294): Setting up sync and frameCommitCallback
D/SurfaceView@aea1e1d( 4294): updateSurfacePosition RenderWorker, frameNr = 1, position = [0, 0, 1080, 2256] surfaceSize = 1080x2256
I/SurfaceView@aea1e1d( 4294): uSP: rtp = Rect(0, 0 - 1080, 2256) rtsw = 1080 rtsh = 2256
I/SurfaceView@aea1e1d( 4294): onSSPAndSRT: pl = 0 pt = 0 sx = 1.0 sy = 1.0
[... cut ...]
I/BluetoothAdapter( 4294): STATE_ON
D/BluetoothLeScanner( 4294): Start Scan with callback
D/BluetoothLeScanner( 4294): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
I/flutter ( 4294): [FBP] <startScan> result: true
I/flutter ( 4294): [FBP] <stopScan> args: null
D/[FBP-Android]( 4294): [FBP] onMethodCall: stopScan
I/BluetoothAdapter( 4294): STATE_ON
I/BluetoothAdapter( 4294): STATE_ON
D/BluetoothLeScanner( 4294): Stop Scan with callback
I/flutter ( 4294): [FBP] <stopScan> result: true

micromotores, 2023-12-28:
I updated the version of flutter to 3.16.5, I have tested it and it is still the same, I copied the log

√  Built build\app\outputs\flutter-apk\app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:56149/h5sxuWo-jw0=/ws
D/[FBP-Android]( 4294): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 4294): [FBP] initializing BluetoothAdapter
I/flutter ( 4294): [FBP] <setLogLevel> args: 5
I/BluetoothAdapter( 4294): BluetoothAdapter() : com.example.last_ble_plus
I/BluetoothAdapter( 4294): STATE_ON
D/[FBP-Android]( 4294): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 4294): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 4294): [FBP] onMethodCall: setLogLevel
I/flutter ( 4294): [FBP] <setLogLevel> result: true
I/flutter ( 4294): [FBP] <getAdapterState> args: null
D/[FBP-Android]( 4294): [FBP] onMethodCall: getAdapterState
I/flutter ( 4294): [FBP] <getAdapterState> result: {adapter_state: 4}
I/ViewRootImpl@de8f851[MainActivity]( 4294): performTraversals params={(0,0)(fillxfill) sim={adjust=resize forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030001
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fl=81810100
I/ViewRootImpl@de8f851[MainActivity]( 4294):   pfl=16020040
I/ViewRootImpl@de8f851[MainActivity]( 4294):   vsysui=500
I/ViewRootImpl@de8f851[MainActivity]( 4294):   apr=LIGHT_STATUS_BARS
I/ViewRootImpl@de8f851[MainActivity]( 4294):   bhv=DEFAULT
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fitSides= naviIconColor=0
I/ViewRootImpl@de8f851[MainActivity]( 4294):   sfl=100000}
I/ViewRootImpl@de8f851[MainActivity]( 4294): performTraversals mFirst=false windowShouldResize=false viewVisibilityChanged=false mForceNextWindowRelayout=false params={(0,0)(fillxfill) sim={adjust=resize forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030001
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fl=81810100
I/ViewRootImpl@de8f851[MainActivity]( 4294):   pfl=16020040
I/ViewRootImpl@de8f851[MainActivity]( 4294):   vsysui=500
I/ViewRootImpl@de8f851[MainActivity]( 4294):   apr=LIGHT_STATUS_BARS
I/ViewRootImpl@de8f851[MainActivity]( 4294):   bhv=DEFAULT
I/ViewRootImpl@de8f851[MainActivity]( 4294):   fitSides= naviIconColor=0
I/ViewRootImpl@de8f851[MainActivity]( 4294):   sfl=100000}
I/ViewRootImpl@de8f851[MainActivity]( 4294): updateBlastSurfaceIfNeeded mBlastBufferQueue=0x72bf0d4a00 isSameSurfaceControl=true
I/BLASTBufferQueue( 4294): update, w= 1080 h= 2400 mName = ViewRootImpl@de8f851[MainActivity] mNativeObject= 0x72bf0d4a00 sc.mNativeObject= 0x724c045160 format= -3 caller= android.view.ViewRootImpl.updateBlastSurfaceIfNeeded:2898 android.view.ViewRootImpl.relayoutWindow:9847 android.view.ViewRootImpl.performTraversals:3884 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301
I/ViewRootImpl@de8f851[MainActivity]( 4294): Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)0 dur=13 res=0x1 s={true 0x724c213000} ch=false seqId=0
I/ViewRootImpl@de8f851[MainActivity]( 4294): updateBoundsLayer: t = android.view.SurfaceControl$Transaction@63ea308 sc = Surface(name=Bounds for - com.example.last_ble_plus/com.example.last_ble_plus.MainActivity@0)/@0xe880aa1 frame = 1
I/ViewRootImpl@de8f851[MainActivity]( 4294): mWNT: t=0x724c109100 mBlastBufferQueue=0x72bf0d4a00 fn= 1 caller= android.view.ViewRootImpl.prepareSurfaces:2985 android.view.ViewRootImpl.performTraversals:4233 android.view.ViewRootImpl.doTraversal:3116
I/ViewRootImpl@de8f851[MainActivity]( 4294): reportNextDraw android.view.ViewRootImpl.performTraversals:4438 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301 android.view.Choreographer$CallbackRecord.run:1309
I/Gralloc4( 4294): mapper 4.x is not supported
W/Gralloc3( 4294): mapper 3.x is not supported
I/gralloc ( 4294): Arm Module v1.0
W/Gralloc4( 4294): allocator 4.x is not supported
W/Gralloc3( 4294): allocator 3.x is not supported
I/BLASTBufferQueue( 4294): [SurfaceView[com.example.last_ble_plus/com.example.last_ble_plus.MainActivity]@0#1](f:0,a:0) onFrameAvailable the first frame is available
D/SurfaceView@aea1e1d( 4294):  setAlpha: mUseAlpha = false alpha=1.0
D/SurfaceView@aea1e1d( 4294):  updateSurfaceAlpha: setUseAlpha() is not called, ignored.
I/ViewRootImpl@de8f851[MainActivity]( 4294): Setup new sync id=0
I/ViewRootImpl@de8f851[MainActivity]( 4294): Setting syncFrameCallback
I/ViewRootImpl@de8f851[MainActivity]( 4294): registerCallbacksForSync syncBuffer=false
I/ViewRootImpl@de8f851[MainActivity]( 4294): mWNT: t=0x724c109400 mBlastBufferQueue=0x72bf0d4a00 fn= 1 caller= android.view.ViewRootImpl.lambda$applyTransactionOnDraw$11$android-view-ViewRootImpl:12865 android.view.ViewRootImpl$$ExternalSyntheticLambda1.onFrameDraw:4 android.graphics.HardwareRenderer$FrameDrawingCallback.onFrameDraw:924
I/ViewRootImpl@de8f851[MainActivity]( 4294): Received frameDrawingCallback syncResult=0 frameNum=1.
I/ViewRootImpl@de8f851[MainActivity]( 4294): Setting up sync and frameCommitCallback
D/SurfaceView@aea1e1d( 4294): updateSurfacePosition RenderWorker, frameNr = 1, position = [0, 0, 1080, 2256] surfaceSize = 1080x2256
I/SurfaceView@aea1e1d( 4294): uSP: rtp = Rect(0, 0 - 1080, 2256) rtsw = 1080 rtsh = 2256
I/SurfaceView@aea1e1d( 4294): onSSPAndSRT: pl = 0 pt = 0 sx = 1.0 sy = 1.0
[... cut ...]
I/BluetoothAdapter( 4294): STATE_ON
D/BluetoothLeScanner( 4294): Start Scan with callback
D/BluetoothLeScanner( 4294): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
I/flutter ( 4294): [FBP] <startScan> result: true
I/flutter ( 4294): [FBP] <stopScan> args: null
D/[FBP-Android]( 4294): [FBP] onMethodCall: stopScan
I/BluetoothAdapter( 4294): STATE_ON
I/BluetoothAdapter( 4294): STATE_ON
D/BluetoothLeScanner( 4294): Stop Scan with callback
I/flutter ( 4294): [FBP] <stopScan> result: true

chipweinberger, 2023-12-28:
please show logs from the example app

micromotores, 2023-12-28:
I/flutter (19668): [FBP] <setLogLevel> args: 5
I/flutter (19668): [FBP] <setLogLevel> result: true
I/flutter (19668): [FBP] <getAdapterState> args: null
I/flutter (19668): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (19668): [FBP] <getSystemDevices> args: null
I/flutter (19668): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (19668): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 8, android_scan_mode: 2, android_uses_fine_location: false}
I/flutter (19668): [FBP] <startScan> result: true
I/flutter (19668): [FBP] <stopScan> args: null
I/flutter (19668): [FBP] <stopScan> result: true

chipweinberger, 2023-12-28:
hmm strange. I would try other FBP versions until you find when the problem started

micromotores, 2023-12-28:
I have tried it with 1.30.9, 1.29.13 and 1.29.0 and the same thing happens with all of them 

Logs:

I/flutter (21316): The Dart VM service is listening on http://127.0.0.1:43815/5hSGYo3d8v0=/
I/flutter (21316): [FBP] <setLogLevel> args: 5
I/flutter (21316): [FBP] <setLogLevel> result: true
I/flutter (21316): [FBP] <getAdapterState> args: null
I/flutter (21316): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (21316): [FBP] <getSystemDevices> args: null
I/flutter (21316): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (21316): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 8, android_scan_mode: 2, android_uses_fine_location: false}
I/flutter (21316): [FBP] <startScan> result: true
I/flutter (21316): [FBP] <stopScan> args: null
I/flutter (21316): [FBP] <stopScan> result: true

chipweinberger, 2023-12-28:
can you search until you find the first version causing a problem? otherwise, I probably cannot fix this.

micromotores, 2023-12-29:
It works for me until version 1.5.2, not after this version.

Do you have any working examples I can refer to?

chipweinberger, 2023-12-29:
try adding the following line in your project/android/app/proguard-rules.pro file:

`-keep class com.lib.flutter_blue_plus.* { *; }`

micromotores, 2023-12-29:
I downloaded the complete example and it worked, thank you. I'm going to adjust my code so it works

closed 2023-12-29 by chipweinberger