chipweinberger/flutter_blue_plus #745: [Bug]: FlutterBluePlus.startScan(timeout: Duration(seconds: 50)) is calling immediately

sureace, 2023-12-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.4

### Flutter Version

3.16.4

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

ESP 32 c3 and ESP 32 WROOM

### What is your problem?

Hi Support Team,
                           When we start the device scan with timeout. It is calling immediately instead of calling after timeout.
FlutterBluePlus.startScan(timeout: Duration(seconds: timeOut)).then((value) {
    debugPrint("Scanning completed");      
});

### Logs

```shell
No Logs
```

chipweinberger, 2023-12-29:
can you try other FBP versions and tell me which version the problem started?

chipweinberger, 2023-12-29:
this is expected. `startScan` returns after scanning *starts*.

you can use 

```
FlutterBluePlus.startScan().timeout(Duration(seconds: timeOut)).then((value) async {
    await FlutterBluePlus.stopScan();
    debugPrint("Scanning completed");
});
```

closed 2023-12-29 by chipweinberger

chipweinberger, 2023-12-29:
I've updated the README