chipweinberger/flutter_blue_plus #746: Get connection parameters [Feature]:

savvn001, 2023-12-29:
### FlutterBluePlus Version

1.31.2

### Flutter Version

3.16.5

### What OS?

All

### OS Version

macOS

### Bluetooth Module

.

### What is your feature request?

It would be nice to have an API to be able to get some other current connection parameters other than MTU for the connected device, for example, connection interval, without having to dig down into the platform APIs. Would this be a fairly straightforward addition?

### Logs

```shell
.
```

chipweinberger, 2023-12-29:
not possible. android and ios don't expose this.

please link me the docs if I know wrong.

savvn001, 2023-12-29:
Oh that's interesting. I just assumed they did. Have you used Nordic's nRF connect app for android? It seems to be able to log when connection parameters are changed, and show information like that. So it's maybe hooking on somehow. But I don't believe that particular app is open sourced.

savvn001, 2024-02-09:
On Android, I think is some kind of callback whenever the connection is updated:

I saw from my logs whilst debugging my app:

`D/BluetoothGatt(13462): onConnectionUpdated() - Device=F0:0D:BE:EF:BB:11 interval=144 latency=0 timeout=400 status=0`

There seems to be this onConnectionUpdated method in BluetoothGatt.java. See this https://stackoverflow.com/a/52757948

chipweinberger, 2024-02-09:
interesting. never saw that before! 

feel free to open a PR

savvn001, 2024-02-11:
Yeah sure. If I get some time I can try to contribute that feature

chipweinberger, 2024-02-12:
I mean, only if you need this feature.

otherwise we shouldn't add it if no one needs it.

elethiame, 2024-07-26:
Modifying the connection interval is crucial for our app. 
Currently, we aim to transmit 100+kb data to the Bluetooth device. However, with a small MTU and an unadjustable connection interval, the user experience is poor.

chipweinberger, 2024-07-26:
modification is not possible

bcliang, 2024-08-01:
> Modifying the connection interval is crucial for our app. Currently, we aim to transmit 100+kb data to the Bluetooth device. However, with a small MTU and an unadjustable connection interval, the user experience is poor.

@elethiame I assume you are using `mtu: 512`.. Did you try using `requestConnectionPriority`? Requesting a high priority connection shrinks the connection interval to 11.25 - 15ms (see https://android.googlesource.com/platform/packages/apps/Bluetooth/+/refs/heads/main/res/values/config.xml#45) -- that is as opposed to the default 30 - 50ms. Still may not speed things up enough.

```dart
device.requestConnectionPriority(
   connectionPriorityRequest: ConnectionPriority.high,
);
```

Going a little deeper, it looks like requesting specific min, max interval, and latency for connection is supported by the `BluetoothGatt` API. I am no expert, though -- perhaps @chipweinberger can provide some clarification.

chipweinberger, 2024-08-02:
can you link to the docs?

it might be possible on android. not ios.

bcliang, 2024-08-02:
This does appear to be android-only: https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int)

Not in the docs, but the underlying supports requesting specific connection parameters.
https://cs.android.com/android/platform/superproject/main/+/main:packages/modules/Bluetooth/framework/java/android/bluetooth/BluetoothGatt.java;drc=4b414a8781b75d173ab28f231890f97bb5d5ce09;l=2042

bcliang, 2024-08-12:
I missed the `@hide` annotation on the `requestLeConnectionUpdate` method. Looks like access is prohibited (at least on modern versions of the Android SDK):
```bash
cat ./hiddenapi-flags.csv | grep leConnectionUpdate
Landroid/bluetooth/IBluetoothGatt$Default;->leConnectionUpdate(ILjava/lang/String;IIIIIILandroid/content/AttributionSource;)V,blocked
Landroid/bluetooth/IBluetoothGatt$Stub$Proxy;->leConnectionUpdate(ILjava/lang/String;IIIIIILandroid/content/AttributionSource;)V,blocked
Landroid/bluetooth/IBluetoothGatt$Stub;->TRANSACTION_leConnectionUpdate:I,lo-prio,max-target-o
Landroid/bluetooth/IBluetoothGatt;->leConnectionUpdate(ILjava/lang/String;IIIIIILandroid/content/AttributionSource;)V,blocked
```

Somewhat related, the **sub-rate** connection methods (`BLE 5.3`) are also not available. That one is more interesting to me.
`cat ./hiddenapi-flags.csv | grep Subrate`