chipweinberger/flutter_blue_plus #749: [Bug]: device.connectionState.listen

hzhh110, 2024-01-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

ex.1.12

### Flutter Version

ex.3.10

### What OS?

All

### OS Version

ex.iOS 12.3

### Bluetooth Module

ESP32

### What is your problem?

device.connectionState.listen
This function triggers multiple disconnected and connected events when connecting to the same device, then disconnecting, then connecting again, then disconnecting again

### Logs

```shell
flutter: linkDevice------
flutter: 22222
flutter: scan OK
flutter: scanDevice
flutter: disconnected :  JH-P130-A0267
2024-01-03 09:23:39.104 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: stopScan
2024-01-03 09:23:39.105 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: startScan
2024-01-03 09:23:39.105 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: connect
flutter: linkDevice------
flutter: 22222
flutter: scan OK
flutter: disconnected :  JH-P130-A0267
2024-01-03 09:23:39.201 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: stopScan
2024-01-03 09:23:39.279 idoor_flutter[2939:37892] [FBP-iOS] didConnectPeripheral
flutter: connected :  JH-P130-A0267
flutter: connected :  JH-P130-A0267
2024-01-03 09:23:39.280 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: connect
2024-01-03 09:23:39.280 idoor_flutter[2939:37892] [FBP-iOS] already connected
2024-01-03 09:23:41.454 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: getSystemDevices
2024-01-03 09:23:41.455 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: disconnect
2024-01-03 09:23:41.455 idoor_flutter[2939:37892] [FBP-iOS] didDisconnectPeripheral:
flutter: disconnected :  JH-P130-A0267
flutter: initTimer3
flutter: state == BleLinkState.BLEOFF
flutter: linkDevice------
flutter: 22222
flutter: scan OK
flutter: scanDevice
flutter: disconnected :  JH-P130-A0267
2024-01-03 09:23:43.239 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: stopScan
2024-01-03 09:23:43.241 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: startScan
2024-01-03 09:23:43.241 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: connect
flutter: linkDevice------
flutter: 22222
flutter: scan OK
flutter: disconnected :  JH-P130-A0267
2024-01-03 09:23:43.336 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: stopScan
2024-01-03 09:23:43.418 idoor_flutter[2939:37892] [FBP-iOS] didConnectPeripheral
flutter: connected :  JH-P130-A0267
flutter: connected :  JH-P130-A0267
flutter: connected :  JH-P130-A0267
2024-01-03 09:23:43.419 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: connect
2024-01-03 09:23:43.419 idoor_flutter[2939:37892] [FBP-iOS] already connected
2024-01-03 09:23:45.788 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: getSystemDevices
2024-01-03 09:23:45.789 idoor_flutter[2939:37892] [FBP-iOS] handleMethodCall: disconnect
2024-01-03 09:23:45.789 idoor_flutter[2939:37892] [FBP-iOS] didDisconnectPeripheral:
flutter: disconnected :  JH-P130-A0267
flutter: disconnected :  JH-P130-A0267
```

hzhh110, 2024-01-03:
I only tested MAC OS

hzhh110, 2024-01-03:
This issue does not occur with iOS

chipweinberger, 2024-01-03:
the problem is that you are listening multiple times.

you need to cancel old subscriptions.

hzhh110, 2024-01-03:
> the problem is that you are probably listening multiple times.
> 
> you need to cancel old subscriptions.

I had to cancel it, but again,macOS has this problem,iOS does not

chipweinberger, 2024-01-03:
iOS and macOS use the exact same code in FBP

hzhh110, 2024-01-03:
Yes, I'm using the same code for iOS and mac

chipweinberger, 2024-01-03:
yes. we use the same code also.

check the problem in the example app

hzhh110, 2024-01-03:
> example app

The example app does not have the option to actively disconnect

chipweinberger, 2024-01-03:
its in the top right corner

https://raw.githubusercontent.com/boskokg/flutter_blue_plus/master/site/example.png

hzhh110, 2024-01-03:
import 'dart:async';
import 'dart:io';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';
enum BleLinkType { DEFAULT, AddDevice, Control }

enum BleLinkState { BLEOFF, SCAN, LINK, LINKING }

class BleTool {
  BleTool._();
  static BleTool get instance => _getInstance();
  static BleTool? _instance;
  var _linkType = BleLinkType.Control;
  int? _detailsId;
  String? _linkName;
  int initTimer = 0;
  BleLinkState _bleLinkState = BleLinkState.LINKING;
  static BleTool _getInstance() {
    _instance ??= BleTool._();
    return _instance!;
  }

  StreamSubscription<BluetoothConnectionState>? _connectionStateSubscription;
  StreamSubscription<List<ScanResult>>? _scanResultsSubscription;
  StreamSubscription<BluetoothAdapterState>? _adapterStateSubscription;
  void chackAdapterState() {}
  Future<void> scanDevice() async {
    print("scanDevice");

    try {
      int divisor = Platform.isAndroid ? 8 : 1;
      await FlutterBluePlus.startScan(
          timeout: const Duration(seconds: 8),
          continuousUpdates: true,
          continuousDivisor: divisor,
          oneByOne: true);
    } catch (e) {
      print("Start Scan Error:${e}");
    }
  }

  void allCancel() {
    if (_scanResultsSubscription != null) {
      _scanResultsSubscription!.cancel();
      _scanResultsSubscription = null;
    }
    if (_adapterStateSubscription != null) {
      _adapterStateSubscription!.cancel();
      _adapterStateSubscription = null;
    }
    if (_connectionStateSubscription != null) {
      _connectionStateSubscription!.cancel();
      _connectionStateSubscription = null;
    }
  }

  void disLinkDevice() {
    var devices = FlutterBluePlus.connectedDevices;
    for (int i = 0; i < devices.length; i++) {
      devices[i].disconnect();
    }
    allCancel();
  }

  void linkDevice(ScanResult scanResult) async {
    print("linkDevice------");
    scanResult.device.connect(mtu: null, autoConnect: false);
    _connectionStateSubscription = scanResult.device.connectionState
        .listen((BluetoothConnectionState state) {
      if (state == BluetoothConnectionState.disconnected) {
        print("disconnected :  ${scanResult.advertisementData.advName}");
      }
      if (state == BluetoothConnectionState.connected) {
        print("connected :  ${scanResult.advertisementData.advName}");
      }
    });
    print("22222");
  }

  Future<void> linkBle(String linkName,
      {int? detailsId,
      BleLinkType linkType = BleLinkType.Control,
      Function(BleLinkState)? linkState}) async {
    _detailsId = detailsId;
    _linkName = linkName;
    _linkType = linkType;
    initTimer++;
    print("initTimer${initTimer}");
    _bleLinkState = BleLinkState.LINKING;
    allCancel();
    if (linkState != null) {
      linkState(FlutterBluePlus.adapterState != BluetoothAdapterState.on
          ? BleLinkState.BLEOFF
          : BleLinkState.LINKING);
    }
    if (FlutterBluePlus.adapterState == BluetoothAdapterState.on) {
      scanDevice();
    }
    _adapterStateSubscription = FlutterBluePlus.adapterState.listen((state) {
      if (state != BluetoothAdapterState.on) {
        if (linkState != null) {
          linkState(BleLinkState.BLEOFF);
        }
      } else {
        scanDevice();
      }
    });

    FlutterBluePlus.scanResults.lastWhere((element) => false);
    _scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      for (int i = 0; i < results.length; i++) {
        if (results[i].advertisementData.advName.endsWith(linkName)) {
          if (linkState != null) {
            linkState(BleLinkState.LINK);
          }
          linkDevice(results[i]);
          print("scan OK");
          FlutterBluePlus.stopScan();
          return;
        }
      }
    }, onError: (e) {});
  }

  Future<void> aaa(BluetoothDevice device) async {
    // Note: You must call discoverServices after every re-connection!
    List<BluetoothService> services = await device.discoverServices();
    services.forEach((service) {
      // do something with service
    });
  }
}

hzhh110, 2024-01-03:
`import 'dart:async';
import 'dart:io';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';
enum BleLinkType { DEFAULT, AddDevice, Control }

enum BleLinkState { BLEOFF, SCAN, LINK, LINKING }

class BleTool {
  BleTool._();
  static BleTool get instance => _getInstance();
  static BleTool? _instance;
  var _linkType = BleLinkType.Control;
  int? _detailsId;
  String? _linkName;
  int initTimer = 0;
  BleLinkState _bleLinkState = BleLinkState.LINKING;
  static BleTool _getInstance() {
    _instance ??= BleTool._();
    return _instance!;
  }

  StreamSubscription<BluetoothConnectionState>? _connectionStateSubscription;
  StreamSubscription<List<ScanResult>>? _scanResultsSubscription;
  StreamSubscription<BluetoothAdapterState>? _adapterStateSubscription;
  void chackAdapterState() {}
  Future<void> scanDevice() async {
    print("scanDevice");

    try {
      int divisor = Platform.isAndroid ? 8 : 1;
      await FlutterBluePlus.startScan(
          timeout: const Duration(seconds: 8),
          continuousUpdates: true,
          continuousDivisor: divisor,
          oneByOne: true);
    } catch (e) {
      print("Start Scan Error:${e}");
    }
  }

  void allCancel() {
    if (_scanResultsSubscription != null) {
      _scanResultsSubscription!.cancel();
      _scanResultsSubscription = null;
    }
    if (_adapterStateSubscription != null) {
      _adapterStateSubscription!.cancel();
      _adapterStateSubscription = null;
    }
    if (_connectionStateSubscription != null) {
      _connectionStateSubscription!.cancel();
      _connectionStateSubscription = null;
    }
  }

  void disLinkDevice() {
    var devices = FlutterBluePlus.connectedDevices;
    for (int i = 0; i < devices.length; i++) {
      devices[i].disconnect();
    }
    allCancel();
  }

  void linkDevice(ScanResult scanResult) async {
    print("linkDevice------");
    scanResult.device.connect(mtu: null, autoConnect: false);
    _connectionStateSubscription = scanResult.device.connectionState
        .listen((BluetoothConnectionState state) {
      if (state == BluetoothConnectionState.disconnected) {
        print("disconnected :  ${scanResult.advertisementData.advName}");
      }
      if (state == BluetoothConnectionState.connected) {
        print("connected :  ${scanResult.advertisementData.advName}");
      }
    });
    print("22222");
  }

  Future<void> linkBle(String linkName,
      {int? detailsId,
      BleLinkType linkType = BleLinkType.Control,
      Function(BleLinkState)? linkState}) async {
    _detailsId = detailsId;
    _linkName = linkName;
    _linkType = linkType;
    initTimer++;
    print("initTimer${initTimer}");
    _bleLinkState = BleLinkState.LINKING;
    allCancel();
    if (linkState != null) {
      linkState(FlutterBluePlus.adapterState != BluetoothAdapterState.on
          ? BleLinkState.BLEOFF
          : BleLinkState.LINKING);
    }
    if (FlutterBluePlus.adapterState == BluetoothAdapterState.on) {
      scanDevice();
    }
    _adapterStateSubscription = FlutterBluePlus.adapterState.listen((state) {
      if (state != BluetoothAdapterState.on) {
        if (linkState != null) {
          linkState(BleLinkState.BLEOFF);
        }
      } else {
        scanDevice();
      }
    });

    FlutterBluePlus.scanResults.lastWhere((element) => false);
    _scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      for (int i = 0; i < results.length; i++) {
        if (results[i].advertisementData.advName.endsWith(linkName)) {
          if (linkState != null) {
            linkState(BleLinkState.LINK);
          }
          linkDevice(results[i]);
          print("scan OK");
          FlutterBluePlus.stopScan();
          return;
        }
      }
    }, onError: (e) {});
  }

  Future<void> aaa(BluetoothDevice device) async {
    // Note: You must call discoverServices after every re-connection!
    List<BluetoothService> services = await device.discoverServices();
    services.forEach((service) {
      // do something with service
    });
  }
}
`

hzhh110, 2024-01-03:
[ble_tool.txt](https://github.com/boskokg/flutter_blue_plus/files/13815083/ble_tool.txt)

hzhh110, 2024-01-03:
> its in the top right corner. see photo:
> 
> https://raw.githubusercontent.com/boskokg/flutter_blue_plus/master/site/example.png

Yeah, I'll try

chipweinberger, 2024-01-03:
its clear in your logs, every time you "link" a device you are creating a new scubscription. this is why every time you are seeing the word "connected" in your logs more and more each time.

you should check if the subscription is null before you recreat it.

`_connectionStateSubscription ??= scanResult.device.connectionState`

chipweinberger, 2024-01-03:
btw, it might be easier to use 

`FlutterBluePlus.events.onConnectionStateChanged`

closed 2024-01-03 by chipweinberger

hzhh110, 2024-01-03:
> its clear in your logs, every time you "link" a device you are creating a new scubscription. this is why you are seeing the word "connected" in your logs more and more each time.
> 
> you should check if the subscription is null before you creat e it.
> 
> `_connectionStateSubscription ??= scanResult.device.connectionState`

Thank you very much!😃